openhardwaremonitor/openhardwaremonitor #733: Nuvoton NCT6791D Sensor Readings Unresponsive after Wake from S3 Sleep State

JTB3, 2015-08-26:
On my windows 8.1 64bit system (ASUS Z87iDeluxe Mobo),
After the computer wakes from an S3 sleep state, the Nuvoton NCT6791D Sensor readings become unresponsive and ouput max values.  Note: All other CPU/sensor respond normally.

Temporary Workaround Solutions:
1) File->Reset (From App Window)
2) Exit and restart app
3) Create a sheduled task using Task Manager using an AHK script to terminate and re-run OpenHardwareMonitor.exe (this is what I ended up doing).

Thanks for creating this awesome utility!!!

moellerm, 2015-12-01:
I think the i/o space lock is enabled again after waking from sleep, so the lines

https://github.com/openhardwaremonitor/openhardwaremonitor/blob/master/Hardware/LPC/LPCIO.cs#L264-L275

should be executed again after waking from sleep. I guess this state is not correctly restored by the waking from sleep function of Windows.

I'm not yet sure what the best fix is for this issue. I don't want to run the above lines of code on every sensor update. So one would have to add code to detect a wake from sleep and the run those lines again.

JTB3, 2015-12-01:
Your suggestion to: "to add code to detect a wake from sleep and the run those lines again." sounds great.  Would love to see this in the next version of OHM!  Thanks for responding. -JT

nchapet, 2015-12-01:
maybe this could help you :

http://stackoverflow.com/questions/9519919/notification-when-windows-enters-in-sleep-mode
http://stackoverflow.com/questions/7253714/get-notified-about-windows-is-going-to-sleep-waking-up-in-c

JTB3, 2015-12-01:
Excellent, Looks like we have the necessary components to address this issue now! (if only I had current C/C++ skills :)

nchapet, 2015-12-02:
I hope too ! :-)

just to notice, there isn't any problem at all with a Fintek F71882 chip. everything work smooth on wake up.

moellerm, 2015-12-04:
This issue should be fixed now. It would be good if someone can test and report back here.

nchapet, 2015-12-04:
it seems to be OK for me, everything run smooth now. you rule !

moellerm, 2015-12-04:
Thx for your quick feedback.

closed 2015-12-04 by moellerm

JTB3, 2015-12-04:
What's the best way to test your update moellerm?  I didn't see a new release update. Can you email or attach a zip for me?

moellerm, 2015-12-06:
I have uploaded an alpha version with the fix here:

http://openhardwaremonitor.org/files/openhardwaremonitor-v0.7.1.1-alpha.zip

JTB3, 2015-12-06:
Michael, Just tested v0.7.1.1-alpha and can confirm that this issue is now fixed!
Thank you very much! :)  -JT