openhardwaremonitor/openhardwaremonitor #738: Scripts for fan control or fan control curve

sa5o, 2015-09-21:
Would it be possible to include an automatic fan control mode, like setting up a custom 
fan curve for PWM fans that would correlate to a particular temperature sensor. For example,
setting a PWM fan that cools the HDD that it responds to HDD temperature and increases the
fan speed as the HDD temperature rises.

Or, alternatively, can a script be written, that would do the same ...response to OHWM sensor
readings and respond accordingly.

BangDroid, 2017-08-01:
Precisely the reason for my visit today. This seems reasonably simple to implement; although I'm no coder.
OHW already has temp plots AND fan control functioning. It's just a matter of linking these. Being able to draw curves would be gravy.

bigpodgurc, 2017-08-01:
Dont you have uefi with fancurve shit every half competent uefi has it

BangDroid, 2017-08-03:
not on a particular pc no, it's not uefi

DarthDestroyer, 2017-08-03:
#996 <- Maybe this PR helps? 

Btw: This issue is about 2 Years old :-P

Chirishman, 2019-08-05:
Actually this request is even older: #98

Also I've got the same need, I've got a server I'm running at home and while OHWM can be used to manually set the fans, the SuperMicro board doesn't seem to actually have any built in fan curve options and tries to just run the fans at 100% all of the time.