openhardwaremonitor/openhardwaremonitor #747: lastest commits broke Nuvoton NCT6791D support

nchapet, 2015-11-23:
as seen on these screenshots : 

Version 0.7.1 Beta :

[image]

and last sources compiled :

[image]

moellerm, 2015-12-01:
Can you save and attach a report (File -> Save Report...) for the version compiled from the latest source?

nchapet, 2015-12-01:
of course ! :-)

here it is.
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/48869/OpenHardwareMonitor.Report.txt)

nchapet, 2015-12-01:
and I want to ask , is there something to do for OHM continues to monitor certain values ​​after waking from sleep ? indeed on 0.7.1 , all values ​​of the chip Nuvoton NCT6791D are no longer monitored, at least until the reset of the software.

moellerm, 2015-12-01:
I think the i/o space lock is enabled again after waking from sleep, so the lines 

https://github.com/openhardwaremonitor/openhardwaremonitor/blob/master/Hardware/LPC/LPCIO.cs#L264-L275

should be executed again after waking from sleep. I guess this state is not correctly restored by the waking from sleep function of Windows.

closed 2015-12-01 by moellerm

moellerm, 2015-12-01:
Please use issue #733 for the problems when waking from sleep. The issue is currently not fixed in the source code.

nchapet, 2015-12-01:
OK ! :-)