openhardwaremonitor/openhardwaremonitor #749: Fans at default/full speed after wake from S3

Sp4iK, 2015-11-25:
If I set my fans to a custom speed percentaje, then I suspend the computer and at wake up the fans are at full speed. I have to set them to 'default' and then back again to manual.

moellerm, 2015-12-04:
Please save a report in OHM (File -> Save Report...) and attach it here. On which hardware/fan do you have the problem?

Sp4iK, 2015-12-04:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/52203/OpenHardwareMonitor.Report.txt)
There you have. Tested on every fan connector of the board with different fans and they all behave the same, so if I have them set to 50% of speed and put the system on sleep and then wake it up they'll be running at 100% although they're manually set.