openhardwaremonitor/openhardwaremonitor #751: UI Blurry when using dpi scaling

calav3ra, 2015-12-01:
The UI of OHM looks blurry when using a dpi scaling on high-res monitors. A fix for that would be to switch away from GDI based drawing.

For c# based applications, this is discussed here:
http://stackoverflow.com/questions/13228185/winforms-high-dpi-blurry-fonts
http://stackoverflow.com/questions/4075802/creating-a-dpi-aware-application

That would be great to have

ElPumpo, 2016-01-05:
I'm running a 1440p monitor and am not experiencing any blurry font / GUIs on this application.
Yes, I'm running at 125% scale and I haven't modified anything.

calav3ra, 2016-01-05:
Weird. Can't replicate today with 125%. O_o

ElPumpo, 2016-01-06:
Happened to me once in my OS; I started up a DPI-aware application and it worked, the following day I started the application yet again and for some weird reason it wasn't scaling. Restarted my computer and issue solved.

Don't really know what causes this issue.

leewi9, 2021-03-12:
I'm using windows 10 4k 32" monitor with dpi scaling at 200%, the main UI window and status bar icon both looks very blurry.

Could you please fix this issue, because it's really annoying with that blurry thing, and I need to use this app everyday..

ScanuNicco, 2021-03-23:
Not sure if this is related but I have been noticing that text gets very blurry if I open it on my 1440p monitor and drag it to my 4k laptop screen. The laptop is a Dell XPS 15 7590 running Windows 10.

fl0v, 2021-03-23:
I think all old applications have issues if you activate upscaling in windows settings.
Try tweaking the properties of the exe file: 'Compatibility' -> 'Change high DPI settings'
**"Use the DPI That's set for my main display when 'I signed in to Windows'"** will cancel the upscaling for that application only.
Also in some cases it helps **"Scaling performed by: 'Application'"**

Basically the application will look bigger or smaller on each display depending on the resolution but it will keep the same quality.

leewi9, 2021-03-23:
@fl0v did it, the main UI window works, but the taskbar icon percentage looks not accurate, i.e. if ram usage is 50%, the ram icon almost filled up, like it's already 80% or 90% ram usage ...