openhardwaremonitor/openhardwaremonitor #755: ghost sensors on Nuvoton NCT6791D

nchapet, 2015-12-04:
is it possible to remove temperatures and fans sensors as pointed on this screen capture ?

they refer to anything on my system.

thanks !
[image]
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/52268/OpenHardwareMonitor.Report.txt)

moellerm, 2015-12-04:
One can create a mainboard specific configuration in the OHM code. But for that one needs to know which fans are really wired on the mainboard (and the label of the connector), how all the voltages are scaled and how the temperatures are used. 

The easier solution is to just hide the sensors you don't need (right click on sensor -> Hide).

closed 2015-12-04 by moellerm

nchapet, 2015-12-04:
indeed I have seen after opening a ticket. I can possibly provide you this information if you wish , that of the Asus Z97I -PLUS I have.