openhardwaremonitor/openhardwaremonitor #756: No fan speeds for motherboard Gigabyte Z97-D3H-CF

Uvik, 2015-12-04:
All readings by OpenHardwareMonitor seem to be correct, however all fan speeds are missing except for the GPUs'. For the record, I have 2 case fans in addition to the CPU fan. Did I miss something? 

A full OpenHardwareMonitor is available as an attachment.

Thanks :)

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/52611/OpenHardwareMonitor.Report.txt)

moellerm, 2015-12-08:
The mainboard seems to use an ITE IT8620E super I/O chip. The Open Hardware Monitor does currently not support this chip. I do not have datasheet for the IT8620E. There is some chance that the chip is similar to the current ITE IT87XX implementation.

nchapet, 2015-12-08:
it seems to have been already solved here by guzzard : https://github.com/guzzard/open-hardware-monitor/commit/24fea5c0460993a91b2873467e1a8bee2a7d389e

moellerm, 2015-12-08:
I have added an implementation in commit 9afb24d based on the code for the ITE IT8772E. As it turns out this is more or less what guzzard did.

A new alpha version can be found here: 

http://openhardwaremonitor.org/files/openhardwaremonitor-v0.7.1.2-alpha.zip

moellerm, 2015-12-13:
Can you please post a report if you have tested the alpha version on a ITE IT8772E system?

ElPumpo, 2016-01-01:
I have a Gigabyte Z97X-UD3H-BK, tried both 0.7.1.2 and 0.7.1, here is log files:

http://pastebin.com/Bn107uhx - v0.9.1.2

http://pastebin.com/Q6mirVYK - v0.9.1

side note: detects my OS as win8 (is w10)

SuperuserMax, 2016-01-13:
@ElPumpo The Windows detectiion issue is already being handles and has already been commited / updated in #766

ElPumpo, 2016-01-13:
@SuperuserMax the funny thing that apparently you haven't noticed is that I started #766

ElPumpo, 2017-02-22:
Should be noted that this isn't in the latest update (0.8, [openhardwaremonitor-v0.8.0.3-alpha.zip](http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip)) you cannot set fan speed, it doesn't work. One of the fans turns off if maunally adjusted.