openhardwaremonitor/openhardwaremonitor #759: Gigabyte H110M-S2H DDR3-CF / Pentium G4400 no temperature

Darius82, 2015-12-11:
Just built a system with these two components and I can see no temps related to the motherboard nor the processor. Only processor load is displayed.

Using openhardwaremonitor on my main system does show me plenty of info.

Also, Realtemp 3.70 does show the processor temperatures.

moellerm, 2015-12-13:
Please post an Open Hardware Monitor report (File -> Save Report...) from your system with the latest alpha version:

http://openhardwaremonitor.org/files/openhardwaremonitor-v0.7.1.2-alpha.zip

You CPU should be supported with this. I suspect that your mainboard uses an ITE IT8628E, which is currently not supported in Open Hardware Monitor.

Darius82, 2015-12-13:
There you go (I can now see all info from the CPU, no info on the Motherboard as you guessed.):

Open Hardware Monitor Report

--------------------------------------------------------------------------------

Version: 0.7.1.2

--------------------------------------------------------------------------------

Common Language Runtime: 4.0.30319.42000
Operating System: Microsoft Windows NT 6.2.9200.0
Process Type: 64-Bit

--------------------------------------------------------------------------------

Sensors

|
+- Gigabyte H110M-S2H DDR3-CF (/mainboard)
|
+- Intel Pentium G4400 (/intelcpu/0)
|  +- Bus Speed      :  100.364  100.364  100.369 (/intelcpu/0/clock/0)
|  +- CPU Core #1    :   802.91   802.91     3312 (/intelcpu/0/clock/1)
|  +- CPU Core #2    :   802.91   802.91  3312.17 (/intelcpu/0/clock/2)
|  +- CPU Core #1    :       22       21       30 (/intelcpu/0/temperature/0)
|  +- CPU Core #2    :       23       23       27 (/intelcpu/0/temperature/1)
|  +- CPU Package    :       24       24       30 (/intelcpu/0/temperature/2)
|  +- CPU Total      :  2.30769  1.53846  51.2821 (/intelcpu/0/load/0)
|  +- CPU Core #1    :  1.53846        0      100 (/intelcpu/0/load/1)
|  +- CPU Core #2    :  3.07692        0  51.2821 (/intelcpu/0/load/2)
|  +- CPU Package    :  1.57654  1.55692  10.0923 (/intelcpu/0/power/0)
|  +- CPU Cores      : 0.146522 0.137015    8.345 (/intelcpu/0/power/1)
|  +- CPU Graphics   : 0.0161793        0 0.108554 (/intelcpu/0/power/2)
|  +- CPU DRAM       : 0.687528 0.603458  1.26885 (/intelcpu/0/power/3)
|
+- Generic Memory (/ram)
|  +- Memory         :  20.8877  20.8877   20.962 (/ram/load/0)
|  +- Used Memory    :   1.6475   1.6475  1.65336 (/ram/data/0)
|  +- Available Memory :  6.23992  6.23406  6.23992 (/ram/data/1)
|
+- Patriot Blast (/hdd/0)
|  +- Temperature    :       30       30       30 (/hdd/0/temperature/0)
|  +- Used Space     :  25.0695  25.0695  25.0695 (/hdd/0/load/0)

--------------------------------------------------------------------------------

Parameters

|
+- Gigabyte H110M-S2H DDR3-CF (/mainboard)
|
+- Intel Pentium G4400 (/intelcpu/0)
|  |
|  +- CPU Core #1 (/intelcpu/0/temperature/0)
|  |  +- TjMax [°C] : 100 : 100
|  |  +- TSlope [°C] : 1 : 1
|  |
|  +- CPU Core #2 (/intelcpu/0/temperature/1)
|  |  +- TjMax [°C] : 100 : 100
|  |  +- TSlope [°C] : 1 : 1
|  |
|  +- CPU Package (/intelcpu/0/temperature/2)
|  |  +- TjMax [°C] : 100 : 100
|  |  +- TSlope [°C] : 1 : 1
|
+- Generic Memory (/ram)
|
+- Patriot Blast (/hdd/0)
|  |
[... 298 lines cut ...]

--------------------------------------------------------------------------------

GenericHarddisk

Drive name: Patriot Blast
Firmware version: SAFM12.0

 ID Description                        Raw Value    Worst Value Thres Physical
 01 Read Error Rate                    000000000000 100   100   50    -       
 09 Power-On Hours (POH)               440000000000 100   100   0     68      
 0C Power Cycle Count                  1E0000000000 100   100   0     30      
 A8 Unknown                            000000000000 100   100   0     -       
 AA Unknown                            5C0100000000 100   100   10    -       
 AD Unknown                            010000000000 100   100   0     -       
 C0 Emergency Retract Cycle Count      100000000000 100   100   0     -       
 C2 Temperature                        1E001E001E00 70    70    30    30      
 DA Unknown                            000000000000 100   100   50    -       
 E7 Temperature                        640000000000 100   100   0     100     
 F1 Total LBAs Written                 2C0000000000 100   100   0     -       

Logical drive name: C:\
Format: NTFS
Total size: 119507251200
Total free space: 89547370496

moellerm, 2015-12-20:
Please test with this new alpha version

http://openhardwaremonitor.org/files/openhardwaremonitor-v0.7.1.4-alpha.zip

and post a new report.

Darius82, 2015-12-20:
I now have motherboard info, thank you:

Open Hardware Monitor Report

--------------------------------------------------------------------------------

Version: 0.7.1.4

--------------------------------------------------------------------------------

Common Language Runtime: 4.0.30319.42000
Operating System: Microsoft Windows NT 6.2.9200.0
Process Type: 64-Bit

--------------------------------------------------------------------------------

Sensors

|
+- Gigabyte H110M-S2H DDR3-CF (/mainboard)
|  |
|  +- ITE IT8628E (/lpc/it8628e)
|  |  +- Voltage #1     :    1.116    1.044    1.128 (/lpc/it8628e/voltage/0)
|  |  +- Voltage #2     :    2.076    2.076    2.076 (/lpc/it8628e/voltage/1)
|  |  +- Voltage #3     :    2.016    2.016    2.016 (/lpc/it8628e/voltage/2)
|  |  +- Voltage #4     :    1.992    1.992    1.992 (/lpc/it8628e/voltage/3)
|  |  +- Voltage #5     :    0.012    0.012    0.012 (/lpc/it8628e/voltage/4)
|  |  +- Voltage #6     :    1.068    1.068    1.068 (/lpc/it8628e/voltage/5)
|  |  +- Voltage #7     :    1.512    1.512    1.512 (/lpc/it8628e/voltage/6)
|  |  +- Standby +3.3V  :     3.36     3.36     3.36 (/lpc/it8628e/voltage/7)
|  |  +- VBat           :     3.12     3.12     3.12 (/lpc/it8628e/voltage/8)
|  |  +- Temperature #1 :       30       30       30 (/lpc/it8628e/temperature/0)
|  |  +- Temperature #2 :       28       28       28 (/lpc/it8628e/temperature/1)
|  |  +- Temperature #3 :       25       25       33 (/lpc/it8628e/temperature/2)
|  |  +- Fan #1         :   1859.5  1607.14   1859.5 (/lpc/it8628e/fan/0)
|  |  +- Fan #2         :   659.18  657.895  659.824 (/lpc/it8628e/fan/1)
|  |  +- Fan Control #1 :                            (/lpc/it8628e/control/0)
|  |  +- Fan Control #2 :                            (/lpc/it8628e/control/1)
|  |  +- Fan Control #3 :                            (/lpc/it8628e/control/2)
|
+- Intel Pentium G4400 (/intelcpu/0)
|  +- Bus Speed      :  100.364  100.364  100.371 (/intelcpu/0/clock/0)
|  +- CPU Core #1    :   802.91   802.91  3312.24 (/intelcpu/0/clock/1)
|  +- CPU Core #2    :   802.91   802.91  3312.24 (/intelcpu/0/clock/2)
|  +- CPU Core #1    :       22       22       31 (/intelcpu/0/temperature/0)
|  +- CPU Core #2    :       23       23       34 (/intelcpu/0/temperature/1)
|  +- CPU Package    :       24       24       34 (/intelcpu/0/temperature/2)
|  +- CPU Total      :  32.8125  32.8125      100 (/intelcpu/0/load/0)
|  +- CPU Core #1    :  35.9375     6.25      100 (/intelcpu/0/load/1)
|  +- CPU Core #2    :  29.6875  29.6875      100 (/intelcpu/0/load/2)
|  +- CPU Package    :  6.73128  6.73128  20.2501 (/intelcpu/0/power/0)
|  +- CPU Cores      :  5.15477  5.15477  18.6428 (/intelcpu/0/power/1)
|  +- CPU Graphics   : 0.0170245        0 0.206948 (/intelcpu/0/power/2)
|  +- CPU DRAM       : 0.876367 0.775187   1.8553 (/intelcpu/0/power/3)
|
+- Generic Memory (/ram)
|  +- Memory         :  23.2931  23.2931  24.2345 (/ram/load/0)
|  +- Used Memory    :  1.83723  1.83723  1.91148 (/ram/data/0)
|  +- Available Memory :  6.05019  5.97595  6.05019 (/ram/data/1)
|
+- Patriot Blast (/hdd/0)
|  +- Temperature    :       30       30       30 (/hdd/0/temperature/0)
|  +- Used Space     :  20.7623  20.7623  20.7623 (/hdd/0/load/0)

--------------------------------------------------------------------------------

Parameters

|
+- Gigabyte H110M-S2H DDR3-CF (/mainboard)
[... 400 lines cut ...]

--------------------------------------------------------------------------------

GenericHarddisk

Drive name: Patriot Blast
Firmware version: SAFM12.0

 ID Description                        Raw Value    Worst Value Thres Physical
 01 Read Error Rate                    000000000000 100   100   50    -       
 09 Power-On Hours (POH)               BE0000000000 100   100   0     190     
 0C Power Cycle Count                  1F0000000000 100   100   0     31      
 A8 Unknown                            000000000000 100   100   0     -       
 AA Unknown                            5C0100000000 100   100   10    -       
 AD Unknown                            010000000000 100   100   0     -       
 C0 Emergency Retract Cycle Count      100000000000 100   100   0     -       
 C2 Temperature                        1E001E001E00 70    70    30    30      
 DA Unknown                            000000000000 100   100   50    -       
 E7 Temperature                        640000000000 100   100   0     100     
 F1 Total LBAs Written                 320000000000 100   100   0     -       

Logical drive name: C:\
Format: NTFS
Total size: 119507251200
Total free space: 94694440960

moellerm, 2015-12-20:
Does it show any mainboard info in the user interface? Could you post a screenshot?

Because from the report it looks like there are no mainboard sensors active.

Darius82, 2015-12-20:
Ofc mate, there you go:

[image]

moellerm, 2015-12-21:
Thank you very much. The GitHub comment styling system "Markdown" has hidden parts of the report you posted as comment (the first part of the tree in the "Sensors" section). And I was confused and looking for a bug in the Open Hardware Monitor software.

closed 2015-12-21 by moellerm