openhardwaremonitor/openhardwaremonitor #766: Detects Win10 as Win8

ElPumpo, 2016-01-01:
Title says all, let's have a look in a exported report (win10):

Operating System: Microsoft Windows NT 6.2.9200.0

This is not true. I'm running Windows 10, and is a issue with this application.
https://en.wikipedia.org/wiki/Windows_NT#Releases

6.2: Windows 8

nlurski, 2016-01-02:
I am on Windows 10 Education and I get the same result. However, looking through the code, it seems as if logging the operating system is just `w.WriteLine(Environment.OSVersion.ToString());` which should be doing everything right.

ElPumpo, 2016-01-05:
Correct, after reading [this](https://social.msdn.microsoft.com/Forums/vstudio/en-US/79d717b3-5005-40d8-8b8c-f68d2e1866ac/environmentosversion-reports-different-versions-for-installed-applications?forum=vbgeneral) and testing myself it seems like not adding the IDs to `app.manifest` the application will report a false `My.Computer.Info.OSVersion` in VB.NET and `Environment.OSVersion.ToString()` too.

Add the following lines to the `app.manifest` and the issue will be solved.

nlurski, 2016-01-05:
I added the lines and submitted a pull request a few days ago. Hopefully it goes through.

ElPumpo, 2016-01-06:
It better.

closed 2016-07-18 by ElPumpo