openhardwaremonitor/openhardwaremonitor #767: Need full reset

antmck2016, 2016-01-02:
I've recently changed HDs and partitions, so the old names no longer apply.  When I hit reset the HDs names stay the same.  It would be nice to be able to reset HD names.  I've lost track of what HD is what, so the temp and used space is useless.  I think when I first installed Open HD Monitor it use the HDs serial numbers - which isn't much use either.  I need to be able to name then C: Windows, D: Media etc.

nlurski, 2016-01-02:
Right clicking on the disk name allows you to rename the disk.

antmck2016, 2016-01-03:
Did you actually read my message?  If I don't know which drive is which, renaming it isn't going to help.

SuperuserMax, 2016-01-13:
Why don't you have a look inside your PC? Maybe the drive's brands are different like Western Digital and Hitachi? Or do you have the same drives with same sizes multiple times?
It would also be a possibility to use another hardware detection program to have a further look at those HDs, wou know, maybe something portable without installation like HWMonitor or something.

antmck2016, 2016-01-13:
All my drives are WD, different sizes.  That doesn't make much difference as there is no way in OHM to reset to default names anyway.  So I have 3 drives that OHM monitors temperatures for, but no way of knowing which drive in OHM is which actual physical drive in my desktop box.  Yes I can rename a drive in OHM by right clicking on it, but again, there is no way in OHM to identify a physical drive.  Before I had C & D on 1 drive, a drive I used for data (H: I think) and a backup drive E.  I think I named the drive in OHM with C,D wrong in the 1st place.  Now I've changed drives and repartitioned.  Well and truly lost track of which drive in OHM represents which physical drive in my system.  Looking at used space in OHM doesn't help either - with different drive sizes and partitions - too confusing.  There needs to be a way of resetting drive names to default at least.  Or am I just missing something here?  Is this still being maintained?  It hasn't updated since December 2014.

moellerm, 2016-01-13:
You can reset names to the default values by editing them (F2, double click or right click -> Rename), deleting the name completely and leaving the edit (Enter or clicking somewhere else).

I'm not sure if the default names help really in your case, but that's the way to reset names. You can also do a complete reset of OHM by exiting it and deleting the OpenHardwareMonitor.config file.

antmck2016, 2016-01-13:
That was helpful - thanks.  Didn't know you could just leave the names blank.