openhardwaremonitor/openhardwaremonitor #770: WMI stops working if GPU is involved

HenrykRichter, 2016-01-10:
I've stumbled into an issue where neither the "Sensor", nor "Hardware" lists within OpenHardwareMonitor's (abbreviated to OHW below) namespace get populated. I suspect a likely reason for this behavior to be that the updates are performed with different privileges. In turn, the WMI properties seem to get unwritable for the privilege level of the running OHW application.

My platform (relevant parts): Windows 10, Intel Core2Quad, NVidia 9800 (MS supplied drivers), OHW started automatically (by it's own menu setting), myself logged in as local user (only one account created on the machine)

Scenario: I poll WMI every 10s to obtain the current system status via Powershell. The information gets forwarded to an LCD display via one of the serial ports.

Reproduction of the issue:
1. To my luck (in this case) the GPU drivers were left in a semi-working state and OHW couldn't get the GPU information. While that was the case, OHW's WMI worked without any complaints. I could run my script for hours.
2. After fixing the GPU driver, OHW could display GPU information in it's main window but I didn't provide any status information anymore via WMI. The namespace was populated but no sensor and hardware data was available. Interestingly for me, the WMI problem was persistent across reboots. 
3. After manually deleting the OHW namespace in WMI, OHW worked for a short while (<1h) before the problem reappeared.
4. Disabling the "GPU" menu setting in OHW fixed the issue for me (with the obvious exception that I cannot display the GPU status on my LCD).

Thanks btw. for OpenHardwareMonitor. Your work saved me a lot of work, which is much appreciated.

drashna, 2016-03-14:
0.6.0 seems to work more reliably than 0.7.1 for what it's worth.

jtc42, 2016-09-07:
Has there been any progress on this? I just installed a discrete GPU and now both the Sensor and Hardware namespaces are unpopulated. Tried disabling the GPU monitoring in OHW but I still get no WMI information.

Annoyingly, 0.6 works, but doesn't support SkyLake, so while I get WMI data, I still can't get my CPU temp.