openhardwaremonitor/openhardwaremonitor #771: Support for Gigabyte X99-UD5 WIFI-CF

downloadingvirus, 2016-01-11:
Hi any chance you look at support for Gigabyte X99-UD5 WIFI-CF mainboards?  There are no sensors at all from mainboard.

Thanks.

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/85643/OpenHardwareMonitor.Report.txt)

ElPumpo, 2016-01-14:
does http://openhardwaremonitor.org/files/openhardwaremonitor-v0.7.1.2-alpha.zip work?

downloadingvirus, 2016-01-15:
Yes I upgraded to the alpha and it has recognized the chip as ITE IT8620E.  The two on board temperature sensors are working,  However the board has 5 pwm fan headers but only shows RPM for 2 and it has picked up 3 fan controls but is not showing any data only a "-"

ElPumpo, 2016-01-15:
Yeah, v0.7.1.2 is unable to control the fans, see #756 for more info.
However, I am unsure about all fans being detected.