openhardwaremonitor/openhardwaremonitor #773: Support for Asus Z97-P board with IT8606E chip

SpiritOffice, 2016-01-14:
I have just bought Z97-P Board by Asus: https://www.asus.com/Motherboards/Z97P/overview/
It uses an unknown chip ITE IT8606E. There is no datasheets on the internet for it.
After i added this chip to source, openhardwaremonitor reads and shows me the values. (the same way as IT8628E)
[image]
All three Fans are OK, these values are the same i can see in bios. The same is for "temperature#2": it is the "mainboard temperature" in bios. I don't know where are the sensors for temp#1 and temp#2, but they seem reasonable for me.
But all voltages are wrong. And i have no idea where i can look for datasheets. The official site does not offer it either.
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/90741/OpenHardwareMonitor.Report.txt)

ElPumpo, 2016-01-15:
Tried v0.9.1.2?

http://openhardwaremonitor.org/files/openhardwaremonitor-v0.7.1.2-alpha.zip

SpiritOffice, 2016-01-15:
> Tried v0.9.1.2?

I don't know where to find v0.9.1.2. 

If it was unclear, i took the newest source from git. 
And yes, i already tested it with 0.7.1.2-alpha.

ElPumpo, 2016-01-15:
> I don't know where to find v0.9.1.2. 
Totally didn't give you a d/l.

Aight, up to the dev to add support I guess.

ElPumpo, 2016-01-16:
http://openhardwaremonitor.org/files/openhardwaremonitor-v0.7.1.4-alpha.zip then?