openhardwaremonitor/openhardwaremonitor #774: OpenHardware Monitor has Completely disappeared

hushpa, 2016-01-18:
so I downloaded theOpenHardware Monitor   - easy - came right up -started playing a game and the computer locked up as expected - surprisingly it was not the cpu overheating  -

so I got the computer going again after a few tries and OpenHardware Monitor
has Completely disappeared - never saw that one before -

apparently I'm going to have to keep  downloading it  till I get this sorted out

:-?

apparently this guy has disappeared as well - at least as far as gmail is concerned
mmoeller@openhardwaremonitor.org.

Slion, 2016-01-30:
So your computer crashes and you used OHM to check the CPU temperature could not be the issue right?

hushpa, 2016-02-01:
had a virus - long gone now tks

Slion, 2016-02-01:
Can you close the ticket then?

closed 2016-02-01 by moellerm