openhardwaremonitor/openhardwaremonitor #775: AMD Phenom II x6 1055T showing 0 C as CPU temperature

Neuen1, 2016-01-22:
Hi, so various temperature monitoring softwares do this and I don't really know why, but CPU temperature is shown as 0 C. Programs like Aida64 or SpeedFan show it correctly.

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/100679/OpenHardwareMonitor.Report.txt)

OzyWizard, 2016-06-25:
Well the problem must be in your system. I have an AMD Phenom II x6 1055T on a Asus Crosshair IV Formula and it works for me. Its clocked and running seti so the system is always at 100%
https://flic.kr/p/JoDQaX

Neuen1, 2016-06-25:
I figured it out some time ago... It was the unleashed mode option in my BIOS that was causing the issue.