openhardwaremonitor/openhardwaremonitor #776: Using command line interface in the new version

navyzet, 2016-01-22:
Hello.
I used this version https://github.com/openhardwaremonitor/openhardwaremonitor/issues/230.
It's very old.
How can I get a new version of the CLI utility?

nobd, 2016-03-15:
Us monitoring folks are in need of this version too.

AndreyCher, 2016-03-27:
I use v230 too, but it no see temperature sensors in new cpu's and MB. This sensors is very importent for me. Can you make at least save the report through windows cmd, from latest version?

wakaru44, 2016-05-08:
+1
A CLI version would be super handy for mass monitoring, scripting, etc.

Belfi, 2016-07-13:
+1 Compile a new version of OpenHardwareMonitorReport please :[ #822

me21, 2017-07-07:
Recompiled with the latest version of OpenHardwareMonitorLib: 
[OpenHardwareMonitorReport.zip](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1130239/OpenHardwareMonitorReport.zip)

me21, 2017-07-07:
Added console version into solution: https://github.com/openhardwaremonitor/openhardwaremonitor/pull/1000
(wow, the 1000th pull request! :))

damascene, 2020-05-01:
@me21 thank you, it's still working in 2020 :)
Any hope to add it to the official packages on your website?
https://openhardwaremonitor.org/

me21, 2020-05-01:
@damascene it's not my website, making a pull request is all I can do.

Looks like @jochenwezel has more recent pull request #1115 and he also has his fork of this repo plus binaries.

damascene, 2020-05-01:
@me21 I meant the project website. Thank you for the link.

evlazarev007, 2025-05-23:
Replacing the OpenHardwareMonitorLib.dll file from the new version helped me.