openhardwaremonitor/openhardwaremonitor #783: Is this still be supported?

antmck2016, 2016-02-17:
It's 2 years since last update.  Is this still being supported or not anymore?

Phyxion, 2016-02-21:
Would be nice if @moellerm added a few admins to at least merge the pull requests (if decent).

DarthDestroyer, 2016-04-06:
Yep @Phyxion, that would be one Solution!
But i hope, that @moellerm would come back, too!

SearchForTheCode, 2016-04-27:
Totally agree, would be a shame..

Besides some important stuff ist already there like the new skylake cpus LGA 1151 which it seems are not working.

DarthDestroyer, 2016-04-29:
I'm new here at github... 
Is it not possible, that anyone fork it and we work together on extensions, so that @moellerm decide afterwards to implement that, when he cames back? 
I'm no Lawyer and i dont know if it is possible with the given License... But it will be the last Solution, when our @moellerm doesnt came back... 

I hope honestly, that everything is ok with @moellerm and he is only busy or not interested to improve this tool! 
Kind Regards!

SearchForTheCode, 2016-04-29:
Of course its open source. Anyone can use this project, create a new one (under the given license). But those persons need the insight like @moellerm  in terms of documentations (intel, amd, the gpu manufacturers) so they can extend new stuff. It is of course not that "easy" if someone has never done stuff in this area. But it could be an option.

We all hope @moellerm  is fine as he has done great work by providing this monitor.

DarthDestroyer, 2016-04-29:
When you look at the Pull Requests and the forks... there are people who know what they do... 

it begins with small extensions and end up at new hardware support! here are some great people with good knowledge... it would be great to stick them together ;-)

im working at voltage support for nvidia gpus at the moment...

SearchForTheCode, 2016-04-29:
Great would be skylake support, as this is more and more becoming usual. But yeah you are absolutely right about the community (i did forget about the pull requests.)

moellerm, 2016-05-26:
I'm fine and back for now. I have been lacking mostly time.

The latest alpha version 

http://openhardwaremonitor.org/files/openhardwaremonitor-v0.7.1.5-alpha.zip

should support all currently known Intel CPUs. If you have a CPU that is not supported, please create a new issue with an attached report.

closed 2016-05-26 by moellerm

antmck2016, 2016-05-26:
Great to have you back.  Thanks for the update.

Phyxion, 2016-05-27:
@moellerm Could you still add a few admins for pull requests, etc?

SearchForTheCode, 2016-05-27:
Great to have you back.. Thats so true :) 

Generally, if you create an "guideline" or "workflow" for integrating new CPUs, i guess this would help a lot, so we can help you out more easily :)

DarthDestroyer, 2016-05-27:
Nice to have you back! Wohooo!!!