openhardwaremonitor/openhardwaremonitor #786: New motherboard request

darkRd, 2016-02-21:
Hello,
My motherboard is ASUS Z170-A and no any info (Voltages, Temperatures, Fans) is displayed for my motherboard.

[image]

DarthDestroyer, 2016-04-29:
please add a report! ;-)

paulqui, 2016-05-27:
Has there been any update to support the  ASUS Z170-A motherboard yet?

skimbro, 2016-07-12:
Open Hardware Monitor Report

--------------------------------------------------------------------------------

Version: 0.7.1.0

--------------------------------------------------------------------------------

Common Language Runtime: 4.0.30319.42000
Operating System: Microsoft Windows NT 6.2.9200.0
Process Type: 64-Bit

--------------------------------------------------------------------------------

Sensors

|
+- ASUS MAXIMUS VIII GENE (/mainboard)
|
+- Intel Core i7-6700K (/intelcpu/0)
|  +- CPU Total      :  16.4266  13.1148  23.4375 (/intelcpu/0/load/0)
|  +- CPU Core #1    :   15.625  5.46875    56.25 (/intelcpu/0/load/1)
|  +- CPU Core #2    :  14.1439  5.48115   34.375 (/intelcpu/0/load/2)
|  +- CPU Core #3    :    18.75     6.25   31.863 (/intelcpu/0/load/3)
|  +- CPU Core #4    :  17.1875  4.63942  30.4688 (/intelcpu/0/load/4)
|
+- Generic Memory (/ram)
|  +- Memory         :  18.4758  18.3724  18.4839 (/ram/load/0)
|  +- Used Memory    :  2.94265  2.92619  2.94394 (/ram/data/0)
|  +- Available Memory :  12.9844  12.9831  13.0009 (/ram/data/1)
|
+- NVIDIA GeForce GTX 980 Ti (/nvidiagpu/0)
|  +- GPU Core       :      135        0      135 (/nvidiagpu/0/clock/0)
|  +- GPU Memory     :      405      405      405 (/nvidiagpu/0/clock/1)
|  +- GPU Shader     :      270        0      270 (/nvidiagpu/0/clock/2)
|  +- GPU Core       :       52       52       53 (/nvidiagpu/0/temperature/0)
|  +- GPU Core       :        1        0        4 (/nvidiagpu/0/load/0)
|  +- GPU Memory Controller :        5        4        6 (/nvidiagpu/0/load/1)
|  +- GPU Video Engine :        0        0        0 (/nvidiagpu/0/load/2)
|  +- GPU Memory     :  5.13204  5.10966  5.15442 (/nvidiagpu/0/load/3)
|  +- GPU            :        0        0        0 (/nvidiagpu/0/fan/0)
|  +- GPU Fan        :        0        0        0 (/nvidiagpu/0/control/0)
|
+- WDC WD5002AALX-00J37A0 (/hdd/0)
|  +- Temperature    :       37       37       37 (/hdd/0/temperature/0)
|  +- Used Space     :  21.1538  21.1538  21.1538 (/hdd/0/load/0)
|
+- Samsung SSD 850 EVO 250GB (/hdd/1)
|  +- Temperature    :       27       27       27 (/hdd/1/temperature/0)
|  +- Used Space     :  90.0711  90.0711  90.0711 (/hdd/1/load/0)
|
+- Samsung SSD 840 PRO Series (/hdd/2)
|  +- Temperature    :       32       32       32 (/hdd/2/temperature/0)
|  +- Used Space     :  61.8387  61.8387  61.8387 (/hdd/2/load/0)

--------------------------------------------------------------------------------

Parameters

|
+- ASUS MAXIMUS VIII GENE (/mainboard)
|
+- Intel Core i7-6700K (/intelcpu/0)
|
+- Generic Memory (/ram)
|
+- NVIDIA GeForce GTX 980 Ti (/nvidiagpu/0)
|
+- WDC WD5002AALX-00J37A0 (/hdd/0)
|  |
[... 762 lines cut ...]
SSDSamsung

Drive name: Samsung SSD 840 PRO Series
Firmware version: DXM06B0Q

 ID Description                        Raw Value    Worst Value Thres Physical
 05 Reallocated Sectors Count          000000000000 100   100   10    -       
 09 Power-On Hours (POH)               332E00000000 97    97    0     11827   
 0C Power Cycle Count                  DB0900000000 97    97    0     2523    
 B1 Wear Leveling Count                D20000000000 94    94    0     210     
 B3 Used Reserved Block Count (Total)  000000000000 100   100   10    0       
 B5 Program Fail Count (Total)         000000000000 100   100   10    0       
 B6 Erase Fail Count (Total)           000000000000 100   100   10    0       
 B7 Runtime Bad Block Total            000000000000 100   100   10    0       
 BB Uncorrectable Error Count          000000000000 100   100   0     0       
 BE Temperature                        200000000000 57    68    0     32      
 C3 ECC Rate                           000000000000 200   200   0     -       
 C7 CRC Error Count                    000000000000 100   100   0     0       
 EB Power Recovery Count               C40000000000 99    99    0     -       
 F1 Total LBA Written                  3EEDD2270400 99    99    0     -       

Logical drive name: C:\
Format: NTFS
Total size: 229875118080
Total free space: 87721721856

DarthDestroyer, 2016-07-13:
1. It helps sometimes to use the latest Version (but i'm sure, that it will not help in this case) 
Available Version right now: 0.7.1.5
(You have to search for that... Im on mobile right now,  sorry) 

2. It would be awesome and helps a lot, when anyone with the same MoBo posts in already existing Issues and the one with another board opens a new Issue...
You have to know, that it needs to be tracked an solved and at the Point right now, i bet, that your Request (last Post) would get lost in Action... :-/

Ah... at Original Poster... It would be awesome too, when you write a usefull Title ;-)