openhardwaremonitor/openhardwaremonitor #787: nVidia gpu core voltage

jazzup, 2016-02-23:
Seems only available for ati gpu.

Can someone with a nvidia gpu tell if they're getting this value?

rmattuschka, 2016-03-26:
No voltages shown

DarthDestroyer, 2016-06-03:
i'm working on it!

it is possible and nearly done. My Fork includes the first Step to this, the implementation of the Current P-State. With that, i am able to track the voltage!
The Test-Application runs fine and i only need to implement it into this. 

"Coming Soon" :)

jazzup, 2016-06-03:
good news

jazzup, 2017-01-27:
Downloaded and compiled your fork.
I can see P-State changing when stressing the gpu, but core voltage remains at 1.5V regardless.

DarthDestroyer, 2017-01-27:
you are correct,  i hadnt the time to get it... 

the voltage is a placeholder right now... 

i will first create a pull about the pstates... its on the way, maybe at the weekend depending on work... 
nice, that you tested it!

jazzup, 2017-02-10:
Thanks for working on it.
I also tested it with 2 old NVidia cards (series 4 and 6) and the pstates worked flawlessly.

DarthDestroyer, 2017-02-10:
i had made some changes recently... how old is your try? 

thanks for testing! :-)

jazzup, 2017-02-12:
Version 0.8.0.2 alpha in the About form, downloaded your fork 16 days ago.

Glad to help.

angelodemon2017, 2018-05-17:
Good afternoon, it turned out to relieve voltage?