openhardwaremonitor/openhardwaremonitor #791: The application was unable to start correctly (0xc000007b). Click OK to close the application.

zeg685, 2016-03-14:
I redownloaded the program, opened it and everything seems to work just fine.
The error didn't show up yesterday and everything was fine until now.
I'm attaching you the whole broken program to investigate if you need.
https://www.dropbox.com/s/75nwxn8hyx0k8aa/openhardwaremonitor-v0.7.1-beta.rar?dl=0

I have windows 7 x64 for refference, if you have more questions please contact me.

Artym1, 2016-03-17:
zeg685.
Your file OpenHardwareMonitor.exe is infected.
https://www.virustotal.com/en/file/cc8688c40ca64c6b154b980c13ff0d36768bb1b89cec967aca636db758c29e6f/analysis/1458204187/
Your should check your computer.

zeg685, 2016-03-17:
Yup you were right, all my executables started to stop working sooner or later.
Reinstalled windows and problem sovled, thanks and sorry for bothering you with this thread without documenting myself before

closed 2016-05-26 by moellerm