openhardwaremonitor/openhardwaremonitor #792: some cores is lost

lidaiyou, 2016-03-18:
My server with 2 CPUs, every CPU with 8 core. so there are 16 cores.
But openhardwaremonitor show 8 cores.

I debug  openhardwaremonitor  find 

coreId = ((apicId >> (int)(threadMaskWith)) - (processorId << (int)(coreMaskWith)));  
  generated   conflict CoreId.
8 coreId is below
CoreId=0  ---
CoreId=0  ---
CoreId=1
CoreId=1
CoreId=9
CoreId=9
CoreId=10
CoreId=10

So ,the CoreId is  conflict , 

SortedDictionary<uint, List<CPUID>> cores = new SortedDictionary<uint, List<CPUID>>();  (CPUGroup.cs);

SortedDictionary removed 4 CPUID because of the  conflict CoreId.

CPU Name=Intel Xeon E5620 and OS is windows 2008 server 64X

DarthDestroyer, 2016-04-06:
Could you add a Report?

moellerm, 2016-05-27:
Looking at the specification of the Intel Xeon E5620 

http://ark.intel.com/products/47925/Intel-Xeon-Processor-E5620-12M-Cache-2_40-GHz-5_86-GTs-Intel-QPI

I see that each CPU has 4 cores (and 8 threads). That makes a total of 8 cores and 16 threads (2 threads per core) for your system with 2 CPUs. In the Windows Task Manager you see the number of threads of your system, and not the number of cores.

If Open Hardware Monitor shows 8 cores, then everything is working fine. If you think something is still wrong, please attach a report (File -> Save Report...) and a description of what is wrong.

closed 2016-05-27 by moellerm