openhardwaremonitor/openhardwaremonitor #793: T-Bal detection in Windows 10

jharneparg, 2016-03-24:
Having upgraded to Windows 10 OHM no longer detects my T-Bal on start up, the issue was not present on Windows 7. OHM launches but all T Bal sensors are missing, as is the T Bal itself in the hardware section. I suspect changes to Windows 10 mean the T Bal cant be contacted at the time that OHM launches,

The situation can be worked around by launching the T Bal navigator software (which amazingly works flawlessly in Windows 10), and while it is running starting a new instance of OHM.

moellerm, 2016-05-26:
Is the ftd2xx.dll file present on your Windows 10 system (it should be in C:\Windows\System32)? 

If it is missing you can try to install the driver http://www.ftdichip.com/Drivers/D2XX.htm for the FTDI chip used on the T-Balancer.

jharneparg, 2016-05-26:
Hi,
Yes ftd2xx.dll is in the windows system32 directory. Both the tbal software and OHM work, but OHM won't detect the tbal on startup.
My current work around is to start the tbal software, and then use the refresh function in OHM to get it to detect the tbal. After that's done OHM functions as normal, and I close the tbal software.
For information I'm setup so that the tbal is communicating over a COM port (VCP) rather than device driver API.
Thanks,
Morgan.

HS34, 2016-12-17:
Hi,
I've got a similar problem. OpenHardwareMonitor doesn't see my T-Balancer.

Workaround : 
 - Delete both peripherals (USB Serial Port (COM3) and USB Serial Converter) and delete their drivers.
 - Install drivers (Windows Server 2003 x64) found on CD.
 - Reset in OpenHardwareMonitor : the T-Balancer appears.
But if I close OpenHardwareMonitor and reopen it, it doesn't detect the T-Balancer anymore.

If I use the driver from http://www.ftdichip.com/Drivers/D2XX.htm instead, OpenHardwareMonitor never detects the T-Balancer.