openhardwaremonitor/openhardwaremonitor #794: AMD Radeon R9 Temperatures

GadgetSteve, 2016-04-07:
Just to let you know that my laptops GPU Core is showing a core temperature of 511.0 degrees C _without it being on fire_ all of the other temperatures are in the range 40-50.

Version 0.7.1 Beta.

christiansblackburn, 2016-04-10:
Hi Gadget Steve,

For whatever it's worth, I have a desktop with an R9 290, generic drivers from AMD and my temperature is always accurate right now I'm idling at 40 Celsius.  That being said I have a different problem where when I RDP in to my computer the GPU temperature dissapears from my taskbar and is a "-" sign until I return home, log in directly, and then I've got my temperature back. 

Thanks,
Christian Blackburn

GadgetSteve, 2016-04-10:
Hi Chrisian, 

I suspect that a part of the problem is that this laptop has two graphics adapters, Intel HG Graphics 4600 & AMD Radeon R9 M265X with auto-switching - having set blender as an application that uses the R9 acceleration _when on mains power_ & starting Blender I get sensible values from open hardware monitor, 45 C instead of 511 C. I also start getting values rather than minuses for the GPU volatage, core & memory clocks. As soon as I close blender it goes back to silly numbers. I will update the ticket accordingly.

If I recall correctly on many systems the GPU is not involved when you RDP in as the screen is disconnected and all graphics operations are going to RAM.

psyke83, 2016-04-21:
The erroneous temperature reading occurs when the dedicated GPU is switched off. I posted a patch for this issue some time ago, but it wasn't followed up by the developers.

https://github.com/openhardwaremonitor/openhardwaremonitor/issues/240

 There might be a cleaner way to implement it, but this quick hack works fine in practice.