openhardwaremonitor/openhardwaremonitor #797: Asus Sabertooth Z170/Intel i7 Skylake - Still no CPU Temp in 7.1.1 Alpha

voddieUK, 2016-05-01:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/244400/OpenHardwareMonitor.Report.txt)

Hello,

Can you please advise if the Skylake should report temps? The previous version (7.1.0) didnt even show "Temperatures" as a sensor.... so thought id try this new Alpha.

As you can see the sensor is picked up now, but I don't see any values?

Appreciate any help in advance,

Voddie

DarthDestroyer, 2016-05-02:
Try: 
http://openhardwaremonitor.org/files/openhardwaremonitor-v0.7.1.4-alpha.zip

voddieUK, 2016-05-02:
[OpenHardwareMonitor.Report-7.1.4.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/244899/OpenHardwareMonitor.Report-7.1.4.txt)

Hi Darth,

Tried 7.1.4 and the results are the same, please see attached report.

Is there anything else I can try?

Kind regards,

Voddie

DarthDestroyer, 2016-05-02:
Ok... Nice, that you have tried it!

You are starting from a Desktop Folder... Have you tried to run the Application as Administrator?
There is a Problem at the Ring0 driver installation

voddieUK, 2016-05-02:
[OpenHardwareMonitor.Report-7.1.4-2.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/245439/OpenHardwareMonitor.Report-7.1.4-2.txt)

Hi Darth,

Bingo!!

Running from my C Drive I see the Temperatures, along with the clock speeds :)

What exactly is the reason for needing to run from a HDD direct? The reason I ask is that I am trying to create a class plugin for Pulseway, so that I can see temps, and be alerted on the go.

Really appreciate your help so far :+1: 

Kind regards,

Marc

DarthDestroyer, 2016-05-02:
There is no need to run from any other Location!

You can run it from the Desktop, too... BUT, you have to run it as Administrator!

When you go to File-Settings => Compatibility => Settings for all Users => Set the "Run as Admin" checkbox to checked and you are fine! ;)

I dont know "Pulseway"...
But what are you "trying to create"?

voddieUK, 2016-05-02:
Ahaa, something so simple :)

And Pulseway - https://www.pulseway.com/remote-management-software

They have yet to support the new Skylake so thought id implement a plugin until they do, using OHM.

Thank you for your help with this, its been driving me mental all weekend!

:+1:

DarthDestroyer, 2016-05-02:
Google said me the same ;)

Makes me curious... What you are planning ;)

voddieUK, 2016-05-02:
Try Pulseway, its free for non-commercial use.... I have the app on my iPhone, which alerts me when certain services go down, or the temps are too high in all my PC's. My Home Automation NUC works fine, can see the CPU, Temp, Usage etc... just my new Skylake doesnt show anything.

They have an API available, but only loads as a sub-item within the particular PC, so not very optimal, but it will work, hopefully :)

DarthDestroyer, 2016-05-02:
Yeah, i had seen something on their page...

But im not very interested in this field, because i have my owm written "hardware overview and inventory tool" ;)

How do you want to get the Data out of OHM into your API / Pulseway Plugin?

SearchForTheCode, 2016-05-19:
The version 7.1.4 helped on an desktop i5 skylake based cpu, but on server side i7 there are still no cpus.   Is this version based on the latest commit (from Dezember 2015) or is 7.1.4 something a user from here "extended"

DarthDestroyer, 2016-05-23:
afaik, it is the latest commit from @moellerm.

i dont know the exact date and it is not listed...

SearchForTheCode, 2016-05-23:
I see. Its a pity. Seems he was on the correct track with skylake.