openhardwaremonitor/openhardwaremonitor #801: Can't use with UWP

MrPing94, 2016-05-19:
When trying to use DLL in UWP project, i can't use the OpenHardwareMonitor.Hardware.Computer.Open function.

System.IO.FileNotFoundException return from DLL and app crashes

MrPing94, 2016-05-19:
Exception thrown: 'System.IO.FileNotFoundException' in OpenHardwareMonitorLib.dll

Additional information: Could not load file or assembly 'System.Management, Version=2.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a' or one of its dependencies. The system cannot find the file specified.

DarthDestroyer, 2016-05-19:
I dont know much about UWP... BUT, OpenHardwareMonitor and also the library needs Admin Privileges.
Have you tried it? (Run as Admin)

On Start, OpenHardwareMonitor will create a .sys-File for the Ring-Driver - do you have write-access in your current AppDir?

Have you tried OpenHardwareMonitor on your Device?
Could you provide a Log-File?

You gave not much information and i dont want to ask god ;)