openhardwaremonitor/openhardwaremonitor #808: Add support for motherboard Z170M MORTAR

lapinus57, 2016-05-30:
[URL=http://www.hostingpics.net/viewer.php?id=929633Capture.png][IMG]http://img11.hostingpics.net/thumbs/mini_929633Capture.png[/IMG][/URL]

[HWMonitor.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/289918/HWMonitor.txt)
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/289919/OpenHardwareMonitor.Report.txt)

skuill, 2016-06-02:
Have the same problem with MSI Z170I GAMING PRO AC (MS-7980) and MSI Z97I AC (MS-7851).

MSI Z170I GAMING PRO AC chip is NUVOTON NCT5563D (ISA A20h).
MSI Z97I AC chip is NUVOTON NCT6779F (ISA A10h).

[image]

Reports:
[OpenHardwareMonitor.Report_098.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/296352/OpenHardwareMonitor.Report_098.txt)
[OpenHardwareMonitor.Report_201.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/296353/OpenHardwareMonitor.Report_201.txt)

DarthDestroyer, 2016-06-03:
@skuill Could you open an issue / board? Only for Handling and Tracking?

@skuill & @lapinus57 No one is forced to help you out and implement your boards! it would be really nice, when you ask peaceful and request nicely! it doesnt handle your problem or solve it... But it would make us happy - that guys, who wants to help! Thank you!

skuill, 2016-06-03:
@DarthDestroyer thanks for response. Copy my message to the new issue.

lapinus57, 2016-06-09:
Boujour 
Serait-il possible de rajouter le support de la carte mère MSI z170 mortar.
elle utilise une puce Nuvoton NCT6793.
Serait-il possible de savoir comment intégrer cette carte dans le code fichier chip, carte mère...
merci d'avance
Benoit MULLER 
PS : désolé pour mon premier post, et pour mon anglais je suis français