openhardwaremonitor/openhardwaremonitor #809: Workflow (Integrating new components)

SearchForTheCode, 2016-06-01:
Hello,

is it possible to create a workflow for integration new components into the hardwaremonitor?

I am asking this because there must be a procedure to integrate new cpu generations like skylake.

If there would be an simple step by step guide, we could help you out more easily with pull requests.

Something like this:

#  Intel
Go to page intel.com , search for "samplesample". Mostly we need register information from xyz..
Add new type to class xyz.

i hope you know what i have in mind.

DarthDestroyer, 2016-06-03:
That would be great! Maybe a "Wiki" or something else...

For now, with only adding identifiers to existing architectures its not really difficult (when you look at the code-diffs) but there is another intel release out soon!

SearchForTheCode, 2016-06-03:
Thats true.. i just found out by that, that skylake servers sould be currently not working..

 case 0x55:		/* Skylake server */
+	    case 0x5e:		/* Skylake */
+	    case 0x8e:		/* Cabylake */
+	    case 0x9e:		/* Cabylake */

This are the identifiers, but OHM only implements those

              case 0x4E:
              case 0x5E: // Intel Core i5, i7 6xxxx LGA1151 (14nm)
                microarchitecture = Microarchitecture.Skylake;
                tjMax = GetTjMaxFromMSR();
                break;

In case i am getting something wrong, just correct me

DarthDestroyer, 2016-06-03:
That is ne next Issue... Cabylake is not defined atm and i bet, that it needs / uses other Sensordata as Skylake...

And you are right, that only 0x5E is supported atm.
0x55 and the others should end up as "unknown"

Do you have access to one of those? The implementation from the 0x55 isnt difficult.
And, i would try to implement the Cabylake too - as Skylake (for testing, if they give results)
If they give results, we can create a new Architekture with the Skylake Sensors...

I dont have the Hardware to test... So, its difficult for me...

SearchForTheCode, 2016-06-06:
Sadly i neither, if any opportunity comes up waht do you need? The OpenHardwareMonitor Report?

I would have guessed that skylake just works like every other model too? Just add it to the enumeration and it should work, or am i wrong? ( I did not see any other changes for obtaining the sensors or reading tjMax)

DarthDestroyer, 2016-06-06:
I wasnt at Source-Code when i wrote this. So is thought, that you could try it ;)

When it is nearly identical, then we could add it to the enum and are done ;)

SearchForTheCode, 2016-06-06:
Hopefully i can find soon an test-System. Not always that easy. This can take some time, but i will write back if i got any news!

SearchForTheCode, 2016-06-10:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/308828/OpenHardwareMonitor.Report.txt)

I got an report. Where do i have to look?

DarthDestroyer, 2016-06-10:
For the CPU Data, it doesnt look bad! 
It is identified as Broadwell. See Microarchitecture in CPU Block. 
What is it really? 

It is funny, that the Mainboard doesnt say anything about the cpu and his own chipset. When you look at the board request, there should stand something like "unknown xyz chipset" 
Serverboards :-/

SearchForTheCode, 2016-06-13:
Oh you are right.. This is broadwell.. sorry about that (Never trust an customer)

Intel S2600CW2R 2xS2011­v3/16xDDR4/2xGbE
Chipsatz: Intel C610

Y CPU Intel XEON E5­2630v4/10x2.2 GHz/25MB/TRAY
• LGA2011­3, 2.20 (3.10) GHz, 25 MB, 8 GT/s
• 10­Kern (10C/20T) • 85 W TDP • Broadwell­EP

I checked the bill. But thats weird. An Intel Xeon should be "detected" without any problems.

DarthDestroyer, 2016-06-13:
what are you currently missing? 
except the board, it doesnt look bad... but, i could have missed something because mobile :-/

btw: are you German?

SearchForTheCode, 2016-06-13:
The one i got the logs from says there are no cpu temperatures available. If you say from the Logs there must be some available, this means i have to check for a possible missunderstanding.  If so i am sorry for the confusing ^^

DarthDestroyer, 2016-06-13:
|  +- CPU Core #1    :       31       30       32 (/intelcpu/0/temperature/0)
|  +- CPU Core #2    :       30       30       31 (/intelcpu/0/temperature/1)
|  +- CPU Core #3    :       31       30       31 (/intelcpu/0/temperature/2)
|  +- CPU Core #4    :       30       30       31 (/intelcpu/0/temperature/3)
|  +- CPU Core #5    :       31       30       31 (/intelcpu/0/temperature/4)
|  +- CPU Core #6    :       31       30       31 (/intelcpu/0/temperature/5)
|  +- CPU Core #7    :       30       30       31 (/intelcpu/0/temperature/6)
|  +- CPU Core #8    :       30       30       31 (/intelcpu/0/temperature/7)
|  +- CPU Core #9    :       30       29       30 (/intelcpu/0/temperature/8)
|  +- CPU Core #10   :       30       30       31 (/intelcpu/0/temperature/9)
|  +- CPU Package    :       37       36       37 (/intelcpu/0/temperature/10)

It looks good, so far ;-)

SearchForTheCode, 2016-06-13:
wtf i am blind.. Now i have an serious talk with that guy!! 

If i get an real skylake problem system i will come back (and double check it before).

Sorry :(

DarthDestroyer, 2016-06-13:
Nevermind! ;-)
Everything is fine! It was a good Conversation,  you are one who would do something instead of only reqeuesting ;-)

SearchForTheCode, 2016-06-13:
Okay :) i try my best to help

willianma, 2017-11-12:
SearchForTheCode gave a great idea. I was with a little spare time here and with a Skylake X i9 7900X 10 core here from a customer of mine and decided to play around with OpenHardwareMonitor source code. But as I am new to the code I couldn't update it to support Skylake X processors. Before I did try something in C but nothing close to make some code improvements to your OpenHardwareMonitor. I couldn't even realize why only CPU core load was showing. (clocks and cpu core temps not shown). At least the cpuid keeps the same  0x5e because it detects the i9-7900X in the processor name, just don't know why it can't gather the temperatures and current clocks. Does anyone have any glimpses of what is happening?

DarthDestroyer, 2017-11-12:
@willianma a report would help here!

if it detects the correct architecture it should display more data! The Name itself isnt a good source of verification...

You use the latest Version? 0.8.3?
Im on mobile, would link if i could :-\

willianma, 2017-11-12:
I just tested on 0.8.3 and also gave same results. Do you have any ideas of where in the code might be the problem? I will try to fix then I can commit later.. If you want to check I have shared a screenshot in this link below: https://1drv.ms/i/s!AooF1nJ83OQfgYA_fLot-0wDRedUUg