openhardwaremonitor/openhardwaremonitor #811: Using .dll in service exe - can get only CPU load data.

MRokas, 2016-06-02:
HDD, CPU, GPU and Mobo is enabled in code and opened.

```
cp.Open()
cp.HDDEnabled = True
cp.FanControllerEnabled = True
cp.GPUEnabled = True
cp.MainboardEnabled = True
cp.CPUEnabled = True
```

However, it only sees sensors for CPU, however clock/temp is zero, and it only gives load information. If I run it manually as exe, I can get GPU data too, including temperature. Running it manually with elevated permissions, allows me to get cpu+mobo+gpu temperatures - but not if it's run as service with elevated permissions.

DarthDestroyer, 2016-06-03:
Without any Code and Report, its really difficult to track your problem!

MRokas, 2016-06-03:
Managed to get it solved, by fixing task in Task Scheduler.. Kinda weird how using admin account isn't enough to run it as admin..

closed 2016-06-03 by MRokas