openhardwaremonitor/openhardwaremonitor #815: Asus Z170I PRO Gaming - Fan Speeds

iamwyza, 2016-06-18:
Would it be possible for OHM to report the fan speeds off of the Asus Z170I Pro Gaming?  I've downloaded 0.7.1.4 Alpha, which thankfully now reports tempuratures, but not fan speeds.

Thanks for your efforts!

DarthDestroyer, 2016-06-18:
Could you try the latest Version?

http://openhardwaremonitor.org/files/openhardwaremonitor-v0.7.1.5-alpha.zip

iamwyza, 2016-06-18:
No dice.
[image]

DarthDestroyer, 2016-06-19:
First: No Screenshot ;-) A Report... But that is my Fault! Click on File - > Report and you will get a textfile ;-)

Second: I think, you are not alone! When i am correct,  then there is already a z170 issue...

iamwyza, 2016-06-19:
Thanks. See attached.
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/322428/OpenHardwareMonitor.Report.txt)

erikwalfridsson, 2016-08-07:
> Second: I think, you are not alone! When i am correct, then there is already a z170 issue... 
Yes, there are several Z170 issues, but isn´t there a seperate chip for the hardware monitoring? I didn´t think that part was integrated into chipset itself. 

It would be good if someone more knowledgeable could clarify this.

I have the same problem with Asus Z170 pro gaming (not Z170**I** pro gaming), and 0.7.1.4 Alpha but I can´t try with 0.7.1.5 Alpha and post my report now , because that computer is taken apart.

DarthDestroyer, 2016-08-07:
For that, we could need the Report ;-)

The Monitoring "chip" is often the same, in a Series of Mainboards. It could be another, but that information is accessible through the report! 

Feel free to look at iamwyza's report, to know what i mean... 

So, for this... We could need your report ;-)

Kind Regards

ewwink, 2016-09-06:
0xD121 = NCT6793D

Callidus80, 2017-04-01:
I have the same issue. MSI Z170A Gaming M5 with Same Chip ID 0xD121. Hope you have time to include in the next release. 
Also : KEEP UP THE CRACKING WORK ! Love Open Hardware Monitor :)
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/887830/OpenHardwareMonitor.Report.txt)