openhardwaremonitor/openhardwaremonitor #817: [Feature Request] Frame rate / FPS

erikiiofph7, 2016-06-19:
The only thing I'm missing is the frame rate, after a gaming session it would be interesting to see what the minimum FPS was.

DarthDestroyer, 2016-06-19:
I think, that will not happen...

You need to fetch DirectX or OpenGL data and that needs libraries that are totally useless for this tool!

I dont know, if ShadowPlay and his FPS are reachable through nvapi but i dont think so...
Im totally out, about AMD/ATI and there Drivers.

On the other Hand, there are a lot of Tools, that will fetch your needs and collects your FPS data!

iNgeon, 2020-04-12:
> 
> 
> The only thing I'm missing is the frame rate, after a gaming session it would be interesting to see what the minimum FPS was.
DarthDestroyer is probably right, but still would be awesome addition.
Would love to the trend graph with fps\frames to do analysis combined with the sensor readings