openhardwaremonitor/openhardwaremonitor #818: Does not show temperature

dgzambrx99, 2016-06-27:
I'm using a Cherry trail mini-pc and it does not display onboard or CPU temp.
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/334047/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2016-06-27:
First, you could use the latest Version => http://openhardwaremonitor.org/files/openhardwaremonitor-v0.7.1.5-alpha.zip

Second, it is unlikly that the latest Version will fix your problem! Your Mainboard isnt supported and i think, it is not mainstream - so, it will not be supported soon!

Your Options:
1. Hope that anyone add your Board
2. Add your Board by yourself

rmattuschka, 2016-06-29:
You must add this 
https://github.com/openhardwaremonitor/openhardwaremonitor/pull/816

DarthDestroyer, 2016-06-29:
This could help, maybe!

You could compile my Fork and run it! => https://github.com/DarthDestroyer/openhardwaremonitor