openhardwaremonitor/openhardwaremonitor #820: Please update nuget package

miltador, 2016-06-28:
https://www.nuget.org/packages/OpenHardwareMonitor

Last published: 2015-03-08

ElPumpo, 2016-07-13:
There is nothing to update...

Wintermute79, 2016-07-13:
NuGet package has versio 0.7.1
Current is 0.7.1.5
For example Intel Skylake CPU temperatures do not show in the NuGet version but are OK with the latest build

DarthDestroyer, 2016-07-13:
It is not possible by anyone else then @moellerm...
Hope and pray, that he does it!

But, you have noticed maybe, that he isnt available very often :-/

ElPumpo, 2016-07-13:
Wait where did you download 0.7.1.5? Latest I know is 0.7.1.2

DarthDestroyer, 2016-07-13:
Two Options for you, buddy... 

1. Search the Issues where it is posted
2. Replace the Number in the Link from the 0.7.1.2 ;-)

DarthDestroyer, 2016-07-13:
Nevermind... Found it at the first Issue i clicked ;-)

http://openhardwaremonitor.org/files/openhardwaremonitor-v0.7.1.5-alpha.zip

ElPumpo, 2016-07-18:
What's new in v0.7.1.5? Fan control still deosn't work for my mobo.

DarthDestroyer, 2016-07-18:
Latest Intels... 

But MoBos are a seperate thing!

ElPumpo, 2016-07-18:
Well I'm refering to #756

ElPumpo, 2016-07-18:
or #771 - same chip afaik

ElPumpo, 2016-07-18:
Oh latest update added basic support for latest Intel 1151 CPUs?