openhardwaremonitor/openhardwaremonitor #826: Radeon RX480 Current Voltage is not shown

ischilling, 2016-07-15:
This bug was 'worse' three Radeon driver versions ago, the current voltage wasn't shown at all. 

*Current status* is, that the voltage isn't shown occasionally, this is related to #825.

[image]

DarthDestroyer, 2016-07-20:
I would prefer, that you create ONE Issue about that... Only for handling!

ischilling, 2016-07-20:
@DarthDestroyer from my point of view, #825 and #826 are two different bugs, from a developer standpoint I understand they might be fixed in one turn... So which one should I delete?

DarthDestroyer, 2016-07-20:
Yeah... You got it ;-) but its not so bad! 

It doesnt matter, what one will stay... 

Regards

ischilling, 2016-07-20:
On request of @DarthDestroyer #826 is herewith closed, #825 remains :+1:

closed 2016-07-20 by ischilling