openhardwaremonitor/openhardwaremonitor #827: [REQUEST] HDD temperature detection

Warriors-Blade, 2016-07-17:
Can we please get HDD temperature sensing added to OHM?

I'm not sure why its missing when that seems to be vital.

unknown14725, 2016-07-24:
Many storage controllers aren't supported by OHM. 

For example the drives connected to the Marvell 91xx controller on my ASUS P8Z68 Deluxe GEN3 motherboard don't show up in OHM (except for used space, ofc).

DarthDestroyer, 2016-07-24:
HDD Temps are supported at some cases and some manufacturers.

There is a massive lag in compatibility!

But you have multiple Options!
1. wait and pray, that anyone will make it possible
2. get in touch with the code and help out! 

If you are stuck at Number 1: you could provide further informations or atleast a report.
With this informations, you had given is it very difficult to say anything about that!

unknown14725, 2016-07-24:
I've been in contact with the developer of Hard Disk Sentinel, and there is WMI support just around the corner. At least for me, this means my frustrations with using OHM for SMART stats logging (health, temperatures, WAF, etc, etc) will end in the near future.

Almost half of my drives/controllers aren't detected by OHM. In my case this means over 15 drives on multiple systems that have to be automatically monitored by other software (in this case, HD Sentinel, which can send emails when there are problems).

It's a real shame, because OHM is super lightweight, and I really like the WMI implementation and it's instance structuring. It makes for very easy parsing of data to external tools (i.e PRTG).

I wished I had more knowledge on the technical aspects concerning OHM hardware detection, and coding it. I'd certainly have the interest and time to give some help with further development and hw compatibility.

unknown14725, 2016-07-24:
Just wanted to add that just because I want to stop using OHM for SMART monitoring, I definitively won't mean I'll stop using it for mobo and GPU sensor logging, though :)

DarthDestroyer, 2016-07-24:
It is a large problem to gather all the informations you need!
every manufacturer brews his own soup! SMART isnt so well documented/regulated in some cases, so you are stuck at a level, that doesnt makes you happy!

You could help at this point, of course... it would be nice!

You could upload a Report and you could explain what Hardware creates nothing / useless data!
I dont think, that it would be a large "problem"!

Regards!

unknown14725, 2016-07-25:
Haha, yeah the never-ending hunt for more and more data to make stats and graphs from!

When summer holidays is over, I'm gonna sit down and make some reports from all the work computers and post them. I've already posted one a little while ago, but there was no response.

https://github.com/openhardwaremonitor/openhardwaremonitor/issues/814

DarthDestroyer, 2016-07-25:
that would be nice! 

yeah... @moellerm isnt very active at the moment, so i try to help him a little bit out and collect informations for him. 
maybe im able to help out by myself ;-)

so, dont hurry... but i think, it should be possible 

thx for your help so far!