openhardwaremonitor/openhardwaremonitor #828: nVidia GPU Temp missing after driver upgrade

bansh3, 2016-07-18:
Hello,

I`m using an Asus laptop with HD integrated graphics and GTX 950M dedicated video card.
With 368.22 nvidia drivers i was able to see the nvdia GPU temperatures, after some driver updates the GPU Temperature feature is missing
 Currently using 07.1.5-alpha version, same with previous version 0.7.1-beta

DarthDestroyer, 2016-07-19:
First: Please add a Report... 

Second: I see two possible Problems... 
1. OHWM doesnt get your dedicated gpu
2. Nvidia deprecated (changed) some Functions <- that needs some Time, but i know, that there are many deprecated planned... :-/

bansh3, 2016-07-19:
Report submited, you`ll notice that temperature is reading can be read from GPU but not shown.

Name: NVIDIA GeForce GTX 950M
Index: 0
Driver Version: 368.81
Driver Branch: r367_00-282

DeviceID: 0x139A10DE
SubSystemID: 0x178D1043
RevisionID: 0xA2
ExtDeviceID: 0x139A

Thermal Settings

 Sensor[0].Controller: GPU_INTERNAL
 Sensor[0].DefaultMinTemp: 0
 Sensor[0].DefaultMaxTemp: 127
 Sensor[0].CurrentTemp: 35
 Sensor[0].Target: GPU

Best regards.

DarthDestroyer, 2016-07-19:
Only @moellerm can receive that Reports... 

It would be awesome, when you upload it here, that i could have a look into!

bansh3, 2016-07-19:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/372280/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2016-07-19:
It does look like i said... 
I will update my Laptop Drivers to check it!

DarthDestroyer, 2016-07-20:
On Desktop, there is no Problem with the latest nvidia driver (368.81)
I will check the Laptop next

DarthDestroyer, 2016-07-25:
I cant reproduce your problem!
I had tested 2 Devices, one with a M GPU, but it works like it should...

rmattuschka, 2016-07-27:
I have wrote my own version thats works for my.
you can try them

[Uploading OpenHardwareMonitorLib.zip…]()

rmattuschka, 2016-07-27:
[Uploading OpenHardwareMonitorLib.zip…]()
Win 10 mit Win 7 Key installieren

rmattuschka, 2016-07-27:
[Uploading OpenHardwareMonitorLib.zip…]()

s193956, 2016-08-27:
same issue. GTX 970, the 372 driver update, unable to get temps

s193956, 2016-08-28:
wierd today it works, i updated some other drivers though

MikeMaximus, 2016-10-01:
I have the same issue with my ASUS laptop and a GTX 960M.  It does report clock rates and load properly when you're playing a game,  but temperature is never shown.  My nVidia drivers are version 372.70.

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/504667/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2016-10-01:
Maybe, it comes from the Energy Options.
The Dedicated Graphics Card is often (on Desktop) not active! That would explain, why it is available in Games!
You could set the Energy Plan to High Power and/or change your GPU behavior in its Tool - if available!
OHM seems not to know, if it is the onboard or the dedicated gpu :-/

MikeMaximus, 2016-10-01:
I think you're right.  If I start OHM while an app is actively using the nVidia GPU (or use the Reset option), it shows all the data properly including temperature. 👍

It must get confused about what data to show when the integrated GPU is the active one at the time it's initialized.

DarthDestroyer, 2016-10-01:
I would say, the problem is, that it doesnt find an active dedicated gpu... 
That would also explain the struggle, when it switches to onbard...

I dont know, if it is fixable -.-

MikeMaximus, 2016-10-01:
It does seem to actually find it even when it's not active.  It shows empty sensor values (except temperature) that start working when it's active.  I attached to a quick screenshot for an example of when the integrated GPU is active.

[image]

It's makes sense that the sensor values aren't available when it's not active, the temperature node being hidden is the odd bit.

Skrapion, 2017-03-17:
That problem is fixed by pull #952.

MikeMaximus, 2017-03-17:
@Skrapion Thanks! Just tested it and it works for me.

Skrapion, 2017-03-17:
@MikeMaximus You're welcome! I don't know if it will help get the pull accepted, but it might be worthwhile to post on the pull page that you've tested it. Cheers!

bvrulez, 2017-06-06:
I have an ACER F5-771G-50RD with a 4GB GDDR5 GEFORCE GTX 950M and I am using HWM 0.8.0 beta. No temperatures are displayed for the GPU, only clocks, load and data.

DarthDestroyer, 2017-06-06:
@bvrulez 

1. Latest Version: http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip
2. Dedicated GPU is not available, when you try to start OHM! => #952 

Solution: Download Latest Version -> Start something that is using your GPU ( or change the Source-Code like the Pull-Request #952) -> Done ;)

mehmed333, 2020-05-20:
Hi everyone, I know this is an old thread but I 'm having similar problem with nVidia gtx 1050 with my notebook, independent of what driver I use. After booting eveyrhing works fine, also during gaming (when nVidia is working). After being in idle for some time it seems like nVidia shuts down completely and intels gou is used. During that time gpu temp reading disappear. Even worse is that seems like BIOS gets confused and fan stops spinning or gets stuck on some speed from before turning nVidia off. I think it i salao related, I used to get BSOD when starting HWmonitor without using nVidia card at the time. After changing settings in nVidia control pannel- 3d rendering card : from automatic to just nVidia I can now open HWmonitor but BSOD still occur sometimes even in idle. I tried dofferent power modes, including high performance but it still happens. Seems to all be related to loosing connection to nVidia card after some time. I tried posting on MEDION (My Notebooks Vendor) and nVidia community page, but still did not get anything helpful.