openhardwaremonitor/openhardwaremonitor #830: UI Looks wrong on High DPI monitors

FluxDelta, 2016-07-20:
I have a high DPI monitor (4K, 15 inch) on my laptop. Unfortunately some items look very wrong when running openhardwaremonitor on my screen:

- List headers cut away 2/3's of the list header - header height seems hardcoded
- Checkboxes too small to click

[image]

FluxDelta, 2016-07-20:
Try scaling the attached image to 50% on a regular  monitor. That is how it looks on a high DPI monitor.

DarthDestroyer, 2016-07-21:
you got it already! 

small high dpi laptops have a bigger font set! (not really, they resize text for better readability, but its longer to write ;-)) 
the text is atleast doubled in size! 

can you check this? 
The setting should be in Displaysettings.
yeah, i know... at that point everything is very small, but it would be nice, if you could do this for reseach! 

Thanks

FluxDelta, 2016-07-21:
Text looks fine. It is set at 250%. However, it is the graphics that causes problems. I have noticed that the following being way too small:

- Header row height (hard coded to 20, clips text as shown in screenshot)
- Icon sizes
- Checkbox sizes
- Column resize cursor

I played around with the header row height and found that this actually fixes it for me (not proud of the code):

      this.treeView
          .GetType()
          .GetField("_columnHeaderHeight", BindingFlags.Public | BindingFlags.NonPublic | BindingFlags.Instance)
          .SetValue(this.treeView, 40);

DarthDestroyer, 2016-07-21:
like i said, if you change it back to normal 100% it would be fine - only very small!

the ui scaling gets only text in the most situations and will create exactly this!

im not so far in ui coding, so it would be necessary to get the ui scaling mode / setting and use it in every control.

and your Setting (40) isnt right ;) When you have a 250% scaling of 20px it should be 50px.

FluxDelta, 2016-07-22:
Ok, now I understand what you are saying. Yes, at 100% things look correctly (but as you wrote, everything is very tiny).

With regards to scaling non-text (like checkboxes), I would imagine that the underlying graphics library is capable of doing so - it seems that most new programs scale nicely. I did have a look at the commits to the TreeViewAdv project and it seems there has been some changes with regards to heading heigh. So maybe updating to a later version would make sense. However, my skills in UI coding are not that great that I would attempt it.

Anyway, thank you for a great tool.

DarthDestroyer, 2016-07-22:
hey buddy... 

you shouldnt close it! Let it open, for further investigation! 
Any solved Problem makes it better... 

Thanks for your help!

dwilhelm, 2016-12-26:
I've updated the TreeViewAdv code to fix some elements mentioned in this issue (in https://github.com/dwilhelm/TreeViewAdv).

I don't know how (or if) you might want to include it in openhardwaremonitor. 

I guess there are a few ways that the change for this issue might be included into OHM, if it's wanted:
- Leave the Aga.Controls 1.7.0.0 copy under External and apply the change there.
- Use a submodule for TreeViewAdv, leave the code as 1.7.0.0 instead of upgrading it, and then apply the fix in the submodule.
- Use a submodule for TreeViewAdv, upgrade it to the most recent version, and then apply the fix.

For my own testing, I did the third option, starting from my earlier PR https://github.com/openhardwaremonitor/openhardwaremonitor/pull/875, to make it easier to keep things separate to send upstream, if needed.

My steps:
- Move TreeViewAdv to a submodule of the openhardwaremonitor repo.
- Create a branch in the new TreeViewAdv repo called "ohm" to track the tweaks made specifically for openhardwaremonitor in commit b18dfb5.
- Upgrade TreeViewAdv to the most recent version, in case it already included a fix for this issue, as FluxDelta mentioned. (I didn't see a change, but there are other improvements so I used it).
- Apply the change for this issue in TreeViewAdv (https://github.com/dwilhelm/TreeViewAdv/commit/14276888099ede07cb7cdaf2ce3bd736c9f96fd4).
- Use the new TreeViewAdv in my fork of openhardwaremonitor.

Anyway, I've attached before and after screenshots from a 240 DPI display, and I've verified that there's no change on a standard 96 DPI display.
[image]
[image]

Please let me know if I could be doing something differently. Thanks!

dwilhelm, 2017-10-31:
Fixed and merged in https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/4

Whotakesmyname, 2019-05-15:
Will there be any further release including this change? It looks helpful. @DarthDestroyer