openhardwaremonitor/openhardwaremonitor #834: Hi i have a problem!

Flipp3rix, 2016-07-27:
In VB using the open hardware monitor lib i can't acces to core clock speed...
i'm using 
`If sensor.SensorType = SensorType.Clock Then
                        Label7.Text = (sensor.Name)
                        Label8.Text = sensor.Value
                    End If`

but the program show me memory clock...how can i get the core clock? Thanks in Advance!

DarthDestroyer, 2016-07-28:
1. You need a Computer! Make it Public, load it at startup...

`   Dim cp As New Computer()
    cp.Open()
    cp.HDDEnabled = True
    cp.RAMEnabled = True
    cp.GPUEnabled = True
    cp.MainboardEnabled = True
    cp.CPUEnabled = True`

Enable / Disable Hardware that you need / not need.

2. Foreach through cp.Hardware

`For Each hardwareitem As OpenHardwareMonitor.Hardware.IHardware In cp.Hardware
hardwareitem.Update()
If hardwareitem.HardwareType = HardwareType.CPU Then
For Each hardwareitemsensor As OpenHardwareMonitor.Hardware.ISensor In hardwareitem.Sensors
If hardwareitemsensor.SensorType = SensorType.Clock Then 
Label7.Text = hardwareitemsensor.Name
Label8.Text = hardwareitemsensor.Value.ToString()
End If
Next
End if
Next
`

3. After using, close your Computer ;)

Flipp3rix, 2016-07-29:
Ok, but when i use this code
```
For Each hw In cp.Hardware

            If hw.HardwareType = HardwareType.GpuAti Then
                hw.Update()
For Each sensor In hw.Sensors
                    If sensor.SensorType = SensorType.Clock Then
                        Label7.Text = (sensor.Name)
                        Label8.Text = sensor.Value
                    End If
Next
End If
Next
```
Label7and8 show me only memory clock of my GPUAti and not core clock...help me and thanks in Advance!

DarthDestroyer, 2016-07-29:
of course... any clock is in your for each, and the memory clock is the last in the row, so... this will be displayed! you override your "Label.Text" everytime with the for each!

You could handle this with:
`If sensor.name.contains('Memory') Then
LabelMemoryClock.Text = sensor.Value.ToString()
End If`

If you want to know, what is in the for each, i would do it so:
(Instead of your 2 labels)
`Console.WriteLine(sensor.name & sensor.value)`
with this, you will get all Names and the corresponding values!

Flipp3rix, 2016-07-29:
Thanks!!!!! It worked!! Thanks so much!!

DarthDestroyer, 2016-08-06:
Could you post the Solution? That the Answer will be available for others?

If your Problem is solved, could you close this Issue?

Regards