openhardwaremonitor/openhardwaremonitor #835: No ram or gpu displayed

633dh98, 2016-08-01:
Running on ubuntu 16.04.1
Displays motherboard fan speeds and temps, cpu clock

does not display any info for GPU (GTX 660), VRAM, or RAM.

Psensor does give GPU temp, use, VRAM, RAM

DarthDestroyer, 2016-08-01:
You have 2 Options, at that Point!

1. i recommend the newest Version! http://openhardwaremonitor.org/files/openhardwaremonitor-v0.7.1.5-alpha.zip

2. it is necessary, that you post a report for further investigation!

Regards!

633dh98, 2016-08-01:
v0.7.1.5-aplha display is no different than v0.7.1-beta display.

What is needed for a report?
[image]

633dh98, 2016-08-01:
I found the report entry in the file menu and submitted one.

633dh98, 2016-08-01:
By the way OpenHardwareMonitor works great for me on the same machine when running Win7HP.  Just FYI.

DarthDestroyer, 2016-08-01:
I recommend, that you post it here! 

The reports, sent through the app is not accessible by anyone... :-/

633dh98, 2016-08-01:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/394402/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2016-08-01:
Okay... 

So... That, what i can read is, that also the report is missing data. 
It should say something about nvidia driver library/api at the bottom of the report. like the amd/ati lib, but it doesnt say something... 
i havent tried it with ubuntu/linux yet, but i suggest, install/update the drivers maybe? 

It could help maybe, when you could also post a win-report. 

Regards

633dh98, 2016-08-01:
One of the first things I did after installing Ubuntu 16.04.1 was to install the Nvidia proprietary driver.

I changed the report name to differentiate it from the Linux report.
[OHM_Win7.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/394472/OHM_Win7.txt)

633dh98, 2016-08-01:
OK, switched from Nvidia 361.42 to 340.96 driver, no change.
Here is new report.
[OHM_Linux_2.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/394510/OHM_Linux_2.txt)

rmattuschka, 2016-08-02:
No implementation for RAM  and Nvidia on Unix systems

DarthDestroyer, 2016-08-02:
yep... it seems so! :-/

633dh98, 2016-08-02:
Thanks.  Is that something missing in OpenHardwareMonitor or something not being provided by Linux?  Just wondering.

DarthDestroyer, 2016-08-02:
I would say, that it is some missing functionality in mono...