openhardwaremonitor/openhardwaremonitor #837: No info from motherboard (no fans, temps och voltages) Asus Z170 pro gaming

erikwalfridsson, 2016-08-06:
First: Thanks for a GREAT piece of software!
I use OpenHardwareMonitor on every computer I own.

I´m a little "blind" on my new motherboard though.
No fans, no voltages, no mobo temps.

Asus Z170 Pro gaming + Core i5 Skylake 6600k

I can see cpu clocks and temps now though, since I started using ver 0.7.1.4 alpha and running it as administrator.

Thanks in advance!

Best regards / Erik Walfridsson, Sweden

DarthDestroyer, 2016-08-06:
There are 3 things "wrong" at your post!
I will begin with the worst... First: You have not posted a Report 
Second: You posted an Issue that already exists!
Third: You havent used the latest Version!

To be honest, this is not a nasty intentioned statement but when you have looked into the Issues => Your MoBo isnt supported at the Moment, a Report could help to get important informations for implementation - only the latest Version, that is tricky and a little bit hidden at the moment ;)

i hope, i could help!
what i recommend: close your Issue and choose one existing Issue with the same MoBo and post your Report there. For developing, it isnt the best, when there are multiple issues open about the same "problem".

Kind Regards from Germany ;)

closed 2016-08-07 by erikwalfridsson

erikwalfridsson, 2016-08-07:
Thanks for your input. I cant´t post a report right now because that computer is taken apart, but I will later if the problem isn´t solved by then.

Kind regards from Sweden

DarthDestroyer, 2016-08-07:
Merci! 

Nice, that you found one! ;-)

If there is really a difference at the Monitoring Chip, feel free to hurt me :-D

Regards