openhardwaremonitor/openhardwaremonitor #838: OHM running as service?

nixx3d, 2016-08-07:
First of all, thanks for keeping OHM up 2 date! it's very much  @appreciated.
For me it's the perfect program to keep track of the state of my renderslaves and the core tool for a GUI I've build for it (more info on that project here > http://m.rckmn.nl/projects/theherder )

With the previous version (6.x) I've noticed that OHM tends to crash when all the RAM/CPU is at maximum (which happens quite often with certain renders).
I recently have updated to the latest version (7.1.0 & 7.1.5 alpha) and hope the crashing will be less.

When it does crash it also destroys the custom config file (with renamed values) and therefor I now use a batch script to check every hour if OHM is still running and reset the config file to the GUI is still working.

I was wondering if there would be an option for OHM to function as a windows service and if this might create a higher stability so it doesn't crash under extreme computing usage?

thanks!

darkbrain-fc, 2021-11-01:
Maybe it's what you're looking for: https://github.com/darkbrain-fc/HardwareSupervisor

c4ceecee, 2024-09-24:
For future reference. I am running OHM as a service using the following PS script for installation:
`#Check script run as administrator
if (!([Security.Principal.WindowsPrincipal][Security.Principal.WindowsIdentity]::GetCurrent()).IsInRole([Security.Principal.WindowsBuiltInRole] "Administrator")) { Start-Process powershell.exe "-NoProfile -ExecutionPolicy Bypass -File `"$PSCommandPath`"" -Verb RunAs; exit }

Set-Location -Path $PSScriptRoot

[Net.ServicePointManager]::SecurityProtocol = [Net.SecurityProtocolType]::Tls12
Invoke-WebRequest -Uri "https://github.com/winsw/winsw/releases/download/v2.8.0/WinSW.NET4.exe" -OutFile "OpenHardwareMonitor-service.exe"

$config = @'
<service>
  <id>OpenHardwareMonitor</id>
  <name>OpenHardwareMonitor</name>
  <description>OpenHardwareMonitor allows for hardware monitoring</description>
  <executable>OpenHardwareMonitor.exe</executable>
  <arguments></arguments>
  <logmode>rotate</logmode>
</service>

'@

Set-Content "OpenHardwareMonitor-Service.xml" $config

Start-Process -WorkingDirectory $PSScriptRoot -FilePath "$($PSScriptRoot)\OpenHardwareMonitor-service.exe" -ArgumentList "install" -NoNewWindow -PassThru -Wait

Write-Host "Installation done"`

One just needs to put the script in the same folder with the executables from this project and run the script as Administrator in Powershell to install. Then start the service through the 'services' tool in Windows.

[Code adapted from minio: https://github.com/minio/minio-service/blob/master/windows/README.md]