openhardwaremonitor/openhardwaremonitor #839: Intel i5 6600K - Missing clocks and temps values

Hertizch, 2016-08-08:
Using version 0.7.1.5 alpha

http://imgur.com/4Lyy1wE

http://imgur.com/wYxRc2G

DarthDestroyer, 2016-08-08:
Hey,

You miss multiple Things at the Moment! Not only your CPU-Data!
Your MoBo isnt supported atm, too!

Btw. Screenshots doesnt help really! It would be better, if you post a Report!

Kind Regards

Hertizch, 2016-08-08:
Haha, I did not even notice that. I was so focused on the CPU. I'll post my report tomorrow :]

Hertizch, 2016-08-13:
Here is the report:
https://drive.google.com/open?id=0B5vPE3K0S33UUEFISjV5R1ZOOWM

DarthDestroyer, 2016-08-13:
You got an error in your report! 

Could you try to run it as Administrator? 
That should fix the cpu Problem! 

But, you should open another Issue for your unsupported MoBo! 

Regards

Hertizch, 2016-08-13:
Yes, it works now :)

closed 2016-08-13 by Hertizch