openhardwaremonitor/openhardwaremonitor #842: intel z x58300

gat45, 2016-08-20:
hello please update for temperature  this  cpu tanks ( cherry trail cr x5 z8300)

DarthDestroyer, 2016-08-20:
Without a Report (File -> Report & Upload it at this Issue), no one is able to help you!

gat45, 2016-08-20:
hello tank s  this

Open Hardware Monitor Report

--------------------------------------------------------------------------------

Version: 0.7.1.0

--------------------------------------------------------------------------------

Common Language Runtime: 4.0.30319.42000
Operating System: Microsoft Windows NT 6.2.9200.0
Process Type: 64-Bit

--------------------------------------------------------------------------------

Sensors

|
+- Cherry Trail CR (/mainboard)
|
+- Intel Atom x5-Z8300 (/intelcpu/0)
|  +- CPU Total      :  58.9844  10.9261      100 (/intelcpu/0/load/0)
|  +- CPU Core #1    :  57.8125    3.125      100 (/intelcpu/0/load/1)
|  +- CPU Core #2    :   46.875        0      100 (/intelcpu/0/load/2)
|  +- CPU Core #3    :   53.125        0      100 (/intelcpu/0/load/3)
|  +- CPU Core #4    :   78.125        0      100 (/intelcpu/0/load/4)
|
+- Generic Memory (/ram)
|  +- Memory         :  45.3063  38.1637  56.5048 (/ram/load/0)
|  +- Used Memory    :  1.77863  1.49823  2.21827 (/ram/data/0)
|  +- Available Memory :  2.14717  1.70754  2.42757 (/ram/data/1)
|
+- Generic Hard Disk (/hdd/0)
|  +- Used Space     :  28.1725  26.7774  28.3665 (/hdd/0/load/0)

--------------------------------------------------------------------------------

Parameters

|
+- Cherry Trail CR (/mainboard)
|
+- Intel Atom x5-Z8300 (/intelcpu/0)
|
+- Generic Memory (/ram)
|
+- Generic Hard Disk (/hdd/0)

--------------------------------------------------------------------------------

Mainboard

SMBIOS Version: 3.0

BIOS Vendor: American Megatrends Inc.
BIOS Version: 5.11

System Manufacturer: Default string
System Name: Default string
System Version: Default string

Mainboard Manufacturer: AMI Corporation
Mainboard Name: Cherry Trail CR
Mainboard Version: Default string

Processor Manufacturer: Intel
Processor Version: Intel(R) Atom(TM) x5-Z8300 CPU @ 1.44GHz
Processor Core Count: 4
Processor Core Enabled: 4
[... 239 lines cut ...]
 000001A2  00000000  005A0000
 00000606  00000000  00000505
 00000611  00000000  03DDC910
 00000639  00000000  0137A904

Microarchitecture: Unknown
Time Stamp Counter Multiplier: 0

--------------------------------------------------------------------------------

AMD Display Library

Status: -1

--------------------------------------------------------------------------------

GenericHarddisk

Drive name: Generic Hard Disk
Firmware version: Unknown

Logical drive name: C:\
Format: NTFS
Total size: 53307502592
Total free space: 38282797056

gat45, 2016-08-20:
hello tank  

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/428367/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2016-08-20:
First of all, you should use the latest Version: http://openhardwaremonitor.org/files/openhardwaremonitor-v0.7.1.5-alpha.zip

But i think, you could use https://github.com/cmoetzing/openhardwaremonitor/tree/airmont instead. He implemented the Airmont Architecture - that "could" help, maybe!

gat45, 2016-08-20:
> First of all, you should use the latest Version:
> http://openhardwaremonitor.org/files/openhardwaremonitor-
> v0.7.1.5-alpha.zip
>
> But i think, you could use https://github.com/cmoetzing/
> openhardwaremonitor/tree/airmont instead. He implemented the Airmont
> Architecture - that "could" help, maybe!

hello  is no found

DarthDestroyer, 2016-08-20:
yeah... open the issue in the browser! your email split the last part of the url!

here is a oneliner:
http://openhardwaremonitor.org/files/openhardwaremonitor-v0.7.1.5-alpha.zip

moellerm, 2016-11-06:
The latest 0.8.0 beta version should support your CPU.

closed 2016-11-06 by moellerm