openhardwaremonitor/openhardwaremonitor #843: reported as virus by Immunet

arnowi, 2016-08-21:
Is anyone else having this problem?
As soon as I download the programme it is quarantined
[image]

DarthDestroyer, 2016-08-21:
First, make sure "where" you Download this from! 

Second, a False warning could be possible, because it sets some things up, that other (bad) tools used in the past! 

When you pay attention to Nr. 1, you are out of danger!

arnowi, 2016-08-22:
Hello Daniel,

I got it from http://openhardwaremonitor.org/downloads/
Basically I just need to know if it's ok to mark it as safe. It didn't
happen before so it was suspicious to get these messages suddenly.

Thanks for the reply.

Best regards,
Arno

DarthDestroyer, 2016-08-22:
Make sure, that your pc is clean and safe! 
There was a report, where someone had a similar Problem. He had a virus in the executable because it got infected. 

The Source and that URL should be safe! 
No one can safely say 100%, but you are not alone ;-)

Kind Regards

akej74, 2016-09-22:
I can confirm that I get a similar problem when starting OpenHardwareMonitor downloaded here:
[http://openhardwaremonitor.org/files/openhardwaremonitor-v0.7.1.5-alpha.zip](url)

I'm using "Trend Micro" and I get the following warning:

**Malicious Behavior Detections**
Malware behaviour blocking, File System
Target: \Device\Harddsik0\DR0
Operation: Create

Maybe a false positive...?

arnowi, 2016-09-22:
> v0.7.1.5-alpha.zip <http://url>

I'd marked it as safe and so far I've noticed no suspicious behaviour.

closed 2016-11-06 by moellerm