openhardwaremonitor/openhardwaremonitor #844: Unable to launch in Win10 x64

CKKwan, 2016-08-22:
Dear All,

I have upgraded to Anniversary Update a week ago, and it was still working fine, until MS pushes another update to me yesterday.

Anyone facing the same issue? Any solution / workaround?

Thanks in advance

DarthDestroyer, 2016-08-22:
You have not described anything about "What happens"!

So, what happens?

Have you tried to run it as Admin? 

Im up to date and it runs fine...

CKKwan, 2016-08-22:
Dear DarthDestroyer, I click on the EXE (regardless of whether I run as administrator) the UAC popup and I grant permission. After that nothing happen. The UI doesn't show up at all.

Sometimes the OpenHardwareMonitor will shows in Task Manager, and I will be able to kill it. If not, then nothing much I can do. If I launch from File Explorer, the file explorer appears hang, if I launch from Start Menu (so call start menu in win10), the start menu will appears hang as well. :(

No trace had been left behind in System Events viewer.

Hopes that help.

DLipsitz, 2016-08-22:
No, you are not the only one this has happened to.

Under Windows 10, there is no option to opt out of updates they feel you must have.  Unfortunately, some of those updates also kill applications.

In the case of this particular app, I had mine set to load with Windows, to monitor memory usage and (most importantly) temperatures.  This had worked just fine until I discovered that the system had updated with the so-called "anniversary update."  The app still loads on startup, but can neither respond to any commands or be terminated.  I have had to change the name of the executable file to stop it from loading in the future.  As there was never a need to install the program in the first place to use it, there is no way other than what I did to remove it from being started, in order to free up memory that is now wasted on a dead app.

According to Microsoft, only "properly installed" applications will be allowed to function, which in translation means it has to be installed in the registry as well as on the computer.  However, experience tells me that may not solve the issues with the "anniversary update," which has even disabled several "properly installed" apps by Microsoft itself, including Flight Simulator.

And the good folks at Microsoft wonder why I refused to "upgrade" to Windows 10 on my production computers.  Until an update for the Open Hardware Monitor app comes out addressing this issue, my advice is to remove it from any and all Windows 10 computers, and to avoid purchasing computers with Windows 10 on them until Microsoft gets their act together.

DarthDestroyer, 2016-08-23:
So... I can only say,  that it runs fine in W10 on multiple Devices! 

Make sure, that you have .net properly installed! 
Run it as Administrator or grant Permission... 

What you can try: Install all Versions of .net!
I does not know the Version that comes with W10, because i develop on these Maschines... 
There are "all in one dot net installer" available to complete this in one step. 

I think, that this might be the Problem, because it seems to be the difference between us! 

Regards

closed 2018-02-04 by CKKwan