openhardwaremonitor/openhardwaremonitor #851: My temperatures aren't showing

rachelrae, 2016-09-18:
For some reason, the temperature for my CPU isn't showing.
I'm trying to play a game, but my computer starts to lag out of nowhere even though it handled the game fine for a couple hours. The CPU may be overheating, but I can't tell since the temperatures aren't showing for the CPU. It could also be my internet that's the problem, but I just want to check.
I don't know very much about computers, but my motherboard is an ASUS H170-Plus D3 if that helps.
My CPU is an Intel i5-6400.
The temperatures for the generic memory section is showing, just not the CPU.

DarthDestroyer, 2016-09-18:
Add a Report, please!

RoelGo, 2018-01-25:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1665683/OpenHardwareMonitor.Report.txt)

Hey, I have the same CPU. Been using OHM for a while now. Just built the latest version and tried it out. I also don't get any temperature data from my CPU. 
I included a report.

Thanks for the amazing project!

DarthDestroyer, 2018-01-25:
https://ark.intel.com/de/products/88185/Intel-Core-i5-6400-Processor-6M-Cache-up-to-3_30-GHz

DarthDestroyer, 2018-01-25:
Your CPU should be fine, with the latest Sourcecode... Did you include some PRs into it?

RoelGo, 2018-01-25:
If by PRs you meant pull requests then no 😄. Just cloned the master branch and built using VS2017.
I also didn't have temperature data on 8.0.0 and 7.4.0.
I did notice that CPUID's latest version of HWMONITOR also doesn't register/show temperature data.

DarthDestroyer, 2018-01-25:
Oha oha... That could be: Had you done an update to the BIOS because of Spectre or Meltdown?

I had seen many Problems from that and Intel said: Dont use it!

That (or something similar) could explain why HWMonitor fails, also...

RoelGo, 2018-01-25:
No but I do have an pretty old bios version since last time I updated my bios I couldn't overclock my bclk anymore. 
I could try updating the bios but that'll have to wait till the weekend. I'll post my results then.