openhardwaremonitor/openhardwaremonitor #856: Support for GEFORCE GTX 1070

ShamekindsMe, 2016-09-28:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/499149/OpenHardwareMonitor.Report.txt)

Thank you +1:

DarthDestroyer, 2016-09-28:
1. Use the latest Version => http://openhardwaremonitor.org/files/openhardwaremonitor-v0.7.1.5-alpha.zip

2. Make sure, you had run it as admin!

3. Have you installed the latest nvidia Drivers?

ShamekindsMe, 2016-09-28:
Thank you for your super fast answer!  Here is a new report made with the latest version.
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/499178/OpenHardwareMonitor.Report.txt)

I confirm that I have the latest Nvidia driver.  Also, the GPU sensors are not available via GPU-Z. Maybe it's a problem with my card.  I am accessing the machine via Remote Desktop, could it be the cause of the problem? Many thanks.
[image]

DarthDestroyer, 2016-09-28:
Yes, RDP is something problematic at that - dont know exactly why...

Could you try it at local?

ShamekindsMe, 2016-09-28:
Unfortunately, the machine is located in France and I’m in Quebec. :)  I will ask someone to test the machine for me tomorrow morning.  Thanks.

ShamekindsMe, 2016-09-29:
Not working even locally.

DarthDestroyer, 2016-09-29:
That is weird! GPUZ doesnt work, too?

ShamekindsMe, 2016-09-29:
No.  I just have 2 sensors in GPU-Z and they both show 0.0MHZ.  Is the GTX 1070 is supposed to be supported?  If so, maybe I have a problem with my driver or my setup?  What do you think?

DarthDestroyer, 2016-09-29:
i dont see a reason why it shouldnt be supported!

moellerm, 2016-11-06:
Remote desktop does not work because the GPU driver is not loaded. Windows uses a "dummy" GPU driver that redirects the desktop to the remote system.

If it does not work locally (and neither does GPU-Z), try installing the latest drivers from NVIDIA and make sure the GTX 1070 is really active.

There are systems where OHM works fine with a GTX 1070, so this is most likely a driver/system configuration issue.

closed 2016-11-06 by moellerm