openhardwaremonitor/openhardwaremonitor #861: Consider adding a co-maintainer to keep OHM up-to-date?

Ivoz, 2016-10-03:
It seems maybe you a lacking time to keep what is a great open source software up with new hardware releases, etc, so maybe an interested co-contributor you could appoint to help with the work, just leaving you to do releases every so often (or some such other similar arrangement)?

akej74, 2016-10-03:
Just a quick comment on updates, I think it would be great if the page http://openhardwaremonitor.org/ could be updated with a link to the alpha-release (latest update is from 2014). As it is now, I believe the only way to find is it by looking at some of the Issues where it has been linked.

DarthDestroyer, 2016-10-03:
Yeah... It was said some times before, but @moellerm isnt very active at the moment!

A Wiki would help us, too...

For the Logs: Latest Version: http://openhardwaremonitor.org/files/openhardwaremonitor-v0.7.1.5-alpha.zip

tds4u, 2016-10-05:
Btw, is it possible to push changes? I ever get error code 403...

rkkoszewski, 2019-10-04:
Is this project still alive? Seems like the original (and only official mantainer) didn't show signs of life since 2018.. Issues and PRs are just piling up. Any information on what is going on?