openhardwaremonitor/openhardwaremonitor #864: wmiobject access via "root/OpenHardwareMonitor" does not work in Windows 10

crysman, 2016-10-06:
We've upgraded one machine to W10pro. Since then my powershell scripts stopped working. Theproblem is that calling

    get-wmiobject -namespace "root/OpenHardwareMonitor" -Class Sensor

does not produce any output. The same script on Win server 2012R2 works just fine.

However, calling

    get-wmiobject -namespace "root/OpenHardwareMonitor" -List

works.

What might be the problem? Any ideas? Thank you.

akej74, 2016-10-10:
Probably unrelated, but I'm running the following with Python successfully on Win 10:
`hwmon = wmi.WMI(namespace="root\OpenHardwareMonitor")
sensors = hwmon.Sensor(["Name", "Parent", "Value", "Identifier"], SensorType="Temperature")`

allstone, 2016-11-23:
Yup, WMI does not work on my PC too. Running latest cumulative updates winver 1607 build 14393. I checked and OpenHardwareMonitor does not write any sensor data under "root/OpenHardwareMonitor" at all, it is empty. Should OpenHardwareMonitor ask and receive some elevated privileges now?

crysman, 2017-01-02:
Please, could anyone dive-in into this?

zXLestath587Xz, 2017-01-04:
I have the same problem after update my Windows 10 x64 Build 10240 to the last Build 14393.576
(https://support.microsoft.com/en-us/help/4004227/windows-10-update-kb3206632)

Everything was working fine before update.

crysman, 2017-02-06:
Any chance of fixing this OpenHardwareMonitor WMI API bug? :/

dwilhelm, 2017-02-06:
I've never looked at the OpenHardwareMonitor WMI info before, so maybe I'm doing something differently, but I don't have any problem getting output on Windows 10 Pro, either with PowerShell or Python.

There's no output when OHM is not running, but that's the same as on Windows 7 for me (again with both PowerShell and Python).

In case it matters, I'm an admin on all test machines.

DarthDestroyer, 2017-02-06:
@dwilhelm said:

1. OHM must run in the background!
2. OHM must be started as admin (but that is already necessary to receive all sensor data)

btw: the latest version is => http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

crysman, 2017-02-07:
OK, I've made further research and tests.
**Conclusion: 0.8.0 works, 0.7.1 doesn't**

I've found that we had been using an older version of OHWM (0.7.1) on one particular W10 machine, which, indeed, is not working properly. 0.7.1 does work, though, without any trouble on Windows Server 2012R2 machines.

On the other hand, the last try witch freshly downloaded 0.8.0 (yes, it is the latest stable now according to the official download page - http://openhardwaremonitor.org/downloads/) has worked now!

Thanks for your time and sorry for misleading info.

closed 2017-02-07 by crysman

Valinwolf, 2019-06-16:
This issue seems to be a problem again in one of the recent updates to windows. Updating to the alpha version seems to have fixed the issue on Windows 10 Pro version 1809 build 17763.503 64-bit.