openhardwaremonitor/openhardwaremonitor #867: Open Hardware Monitor doesn't show all sensors on Ubuntu 16.04 LTS

t0msk, 2016-10-20:
Screenshot: [http://i.stack.imgur.com/Mgtk8.png](http://i.stack.imgur.com/Mgtk8.png)

I downloaded Open Hardware Monitor and installed mono (to run Open Hardware Monitor) and I installed lm-sensors too. If I run Open Hardware Monitor it shows only fans and voltage, but It doesn't show current CPU clock (and core temperatures), RAM, iGPU / GPU clock / temperature, Discs temperature and SMART info.

In Windows I have no problem. Please can you help me?

t0msk, 2016-11-04:
Can you help me please?

DarthDestroyer, 2016-11-04:
#835

t0msk, 2016-11-04:
So OHM cannot display RAM and other things in linux? So this screenshot on the main page is fake? http://openhardwaremonitor.org/screenshots/

DarthDestroyer, 2016-11-04:
Do you see the Values? I would say, no!

So, i dont know where the Problem comes from!
Maybe, it is related to Mono? 

The Screenshots are real, but they are very very old! 
I would say: Try to figure it out by yourself, share it with the community and everyone will be happy! :)

t0msk, 2016-11-05:
I tried lot of things.. how can i share it with community? I shared it here on github :) So according to you it is problem with mono? Ok i can send them a message about that.

And what is development status of OHM? Because main website is very very outdated.

DarthDestroyer, 2016-11-05:
Github, this Repo is a Community ;-)
Share your Tests and Results here, in your Issue! 

I dont say, it is related to mono... But Maybe a Source of the Problem... Or Ubuntu... There are many Possibilities! 
I cant try anything, i dont have an Ubuntu with gui... 

OHM has a good Community! The Main-Dev isnt really active at the Moment, but you can fork it and make your own... Like i did! :-)

The Main-Page is from the Time before Github... And a little bit useless now ;-)

t0msk, 2016-11-05:
Yes and what is the newest, most updated and the best branch of OHM? Bceuase i downloaded it from website so it can be outdated. I dont think that problem is in Ubuntu, because mono handles the program :)

DarthDestroyer, 2016-11-05:
Look at the Issue, i had posted above! There is a Link to the latest Version! 

Maybe, Ubuntu or Linux is a Problem, too! If they changed the Kernel or Permission at the Sensors or anything related, it will get no data! 
There are more possible Problems, as i can adress from here! 

Go to OHM Git, Go to Network and look on the graph, who has the latest Addition 😉
Im on Mobile... Cant tell you the way, but you will find it ;-)

t0msk, 2016-11-05:
It looks that your fork is most active. But does forks merge to 1 main branch sometimes, like master for new release?

DarthDestroyer, 2016-11-05:
Everything, i had added to my fork is from a pull request at the main! 
That, what i added, got a pull request, too! 

So, if @moellerm love these pulls as i do, it will be the same! 

If not, my fork will be the supported parallel! :-)

t0msk, 2016-11-05:
Ok so main master has all updates from your fork? :) who has access to merge to master? Is he active? :)

DarthDestroyer, 2016-11-05:
Far away from my Fork!
OHM Main is lacking some minor improvements, at the moment!

@moellerm is the one and only and he isnt really active!

t0msk, 2016-11-05:
I think this is quite bad git management, because everyone is creating own fork... it should be 1 "main" fork and others people should create branches from it... Because now there are lot of forks, but there is no branch with all changes (because @moellerm doesnt merge changes to master)

Because i would like to download newest OHM, but i cant because everybody has different things implemented.

DarthDestroyer, 2016-11-06:
yep... that is correct! i dont like the situation, too... But no one else has access and @moellerm is the Master!
You can Fork it, pull requests and hope, that it will be added someday...
I had started the idea of collecting as many pull requests as possible... but it is a lot of work and everyone pulls to the main (it is a good idea... but i have to look every line and change it at my fork)

We can only hope and pray, that it will change something... but i am not at that point to believe that :/

moellerm, 2016-11-06:
The hardware support in OHM under Linux is very limited. On Linux we do (not) load our own kernel driver to access the hardware, but rely on libraries like lm-sensors and gpu drivers to get the data. Only what is exposed by theses drivers is shown in OHM. And even for this the support in OHM is incomplete. The screenshots on the website are not fake, but the results depend a lot on your hardware. 

Only the user interface of OHM uses shared code on Windows and Linux, the rest has to be rewritten from scratch (and the Windows code is currently more complete).

Create pull requests for code that improves the support on Linux.

t0msk, 2016-11-06:
Hi thanks for response, but can i have this issue opened and i will share my results & tests here :)

hirschmann, 2016-11-17:
I could create a PR which adds MSR support on Linux. It would be based on this commit: https://github.com/hirschmann/nbfc/commit/9f4e1510ea3b6213fd2137d4fca49516bd6c9d39

@moellerm 
If you think this could be useful to OHM, please let me know :)

t0msk, 2016-11-17:
@hirschmann what is PR and MSR? thank you very much for that commit, are you a linux user? Did you test it?

hirschmann, 2016-11-20:
PR = pull request
MSR = model specific register (see https://en.wikipedia.org/wiki/Model-specific_register)

Yes, of course I tested it. It's part of my own project.

t0msk, 2016-11-21:
@hirschmann so MSR is support only for CPUs right? Yes you can create PR, i hope that @moellerm will approve it. Is your project about linux integrating?

t0msk, 2016-12-02:
@moellerm what is your opinion to https://github.com/hirschmann/nbfc/commit/9f4e1510ea3b6213fd2137d4fca49516bd6c9d39 ? @

theel0ja, 2017-02-25:
Same problem, using Linux Mint 18.1 here...
[image]

DarthDestroyer, 2017-02-25:
Please Add Reports :-)

t0msk, 2017-03-05:
Do you mean txt files?