openhardwaremonitor/openhardwaremonitor #868: Critical: OHM causes full fan speed on some systems

SearchForTheCode, 2016-10-26:
Hello dear OHM Team,

sadly i have to report an rather critical issue. It is regarding a SuperMicro Server (Swyx Express X20). On this if i start the OHM the fan RPM is getting crazy. The fan immediatly rotates at full speed, like the access to the fan is done incorrectly. I have checked it and only OHM is causing this. If i restart the system, start "speedfan", everything is okay, but as soon as i use OHM the fans goes crazy.. (its no faulty value, you can here the fan go from low to high, from low to high rotation , the moment OHM is started and it never ends)

I have never seen such an issue before with other systems (even other supermicro systems), but as this specific system is sold by swyx as an appliance, this occurs on every system.

I attached an report and hopefully you can find any information in it leading to this RPM. 

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/552538/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2016-10-26:
Is there any other Tool running with OHM?
You said speedfan... Is it open, too?
Have you tried OHM without any other Tool? For Example, AI-Suite - speedfan oder something like that?
Have you checked the Speed-Value at the Fan-Control by Rightclick? Is it set on default?

You can see high RPMs in the Report, what Type of Fan is it? How high is the max-rpm?

SearchForTheCode, 2016-10-27:
At this specific moment yes, speedfan was running, but just to compare RPM Values. For this problem to occure just OHM is running. No i did not check the speed value for "default". (but as this also happens if i use the LIB and not the GUI Tool, this does not seem relevent).

It seems like a normal fan inside an mini case  (i know those RPMs seems crazy high). Normal RPM is 7000-9000RPM and when i start OHM the RPM immediatly flickers between 7000 and those 160000RPM. 160000RPM seems to be the max-rpm. And thats why you "hear" the fan working because 1 second it is in 7000RPM and the other second in 160000RPM. If i close OHM and restart it is gone forever till i start OHM again.

DarthDestroyer, 2016-10-27:
Have you used or tried the gui before? 
Maybe, im not sure at the settings saving, you set inngui another value for example 100% and the lib uses the same value, too? 
Could you check it in the gui? 

How and why are you using the lib? Are you sure, that you havent done anything with the fan-control? ;-)

SearchForTheCode, 2016-10-27:
Nope..I used the GUI just to check if there is an bug in my application using the OHM Lib or if it happens generally in OHM. 

But no i have never used the GUI on those systems. I am using the lib because i integrated the values in another application (where other stuff is also displayed next to the CPU Temps. 

Also note this happens only on this specific system (no other system before). 
Actually i am doing nothing at all with the fans. Just initializing the Computer Objects , collect the sensor data and call "Update". And if i need an specific value i access the value. I did not even know that there is some kind of "fan control" but that is good to know. 

If there is any need for my code part, i can append it (its not many lines, about 200).

DarthDestroyer, 2016-10-27:
Is there a difference between the Systems? Mainboard, CPU? Chipset? 

yeah, every rpm item should be controlable! That seems to be the reason! 

you say, that you init the Computer... what parts do you enable? 
Could you disable them in steps to check, which one makes the problem? 

Maybe, try to set the control to default? Everytime you update? Atleast a try!

SearchForTheCode, 2016-10-27:
The problem is, that i probably wont get any access to the system anymore, but i need to try. 
Yeas i guess there is a difference. I think nearly everything will be different, but i tested on several hundret systems over time and only this one has to have this issue. 

Mainboard is from SuperMicro. 

I enabled this

local.CPUEnabled = true;
  local.FanControllerEnabled = true;
 local.HDDEnabled = true;
 local.MainboardEnabled = true;

if i had to guess i think the problem should come with fancontrollerEnabled, but if i disable it i probably wont get any fans (or am i wrong and i just can disable this)

DarthDestroyer, 2016-10-27:
That is a good question... ;-)
You should try it, i think! 
When im at home, i could try it...

SearchForTheCode, 2016-10-27:
I will, thanks for the info so far!

SearchForTheCode, 2016-10-28:
I tested it. So i guess this is it!  If i disable this flag , i still get the fan information so hopefully just the "control" is disabled. :) Of course i further need to test this, but it seems like a possible solution in this case.

DarthDestroyer, 2016-11-04:
Nice to know! 

Atleast, we found a Problem at the Control Part!

moellerm, 2016-11-06:
The flag Computer.FanControllerEnabled controls if specific fan controller should be enumerated (and monitored). Currently this applies only to two old fan controllers (Heatmaster and TBalancer) connected via virtual serial port (USB). The flag Computer.FanControllerEnabled does not affect any other fan control features present on the mainboard or graphics card.
The report you posted does not show the presence of a Heatmaster or TBalancer. Maybe there is some other device connected to the serial port that resembles a Heatmaster or TBalancer and triggers the behavior. It would be interesting if you can really reproduce the problem (multiple times) with Computer.FanControllerEnabled = true and it does not show (multiple times) when Computer.FanControllerEnabled = false.

I would suspect that the problem is caused by the access to the Winbond W83627DHG-P chip present on the mainboard, because the CPU fan should be controlled by this chip as well. You could disable monitoring of the Winbond W83627DHG-P by setting Computer.MainboardEnabled = false. If the problem is really caused by the access to the Winbond W83627DHG-P, then one has to start searching for the problem in these lines:

https://github.com/openhardwaremonitor/openhardwaremonitor/blob/master/Hardware/LPC/W836XX.cs#L262

SearchForTheCode, 2016-11-07:
Okay, that is "sad" to hear, from an analysis standpoint as i thought fancontrollerenabled could be the solution. I will test tomorrow (i only have this one test sadly) several configs. One with fanControllerEnabled=false and the other one with mainboardEnabled=false and will report back to you. 

Any further feedback i could need or i should pay attention too?

moellerm, 2016-11-07:
Please save reports (File -> Save Report...) for all situations tested (where the problem occurs and where it does not occur) and attach them here.

SearchForTheCode, 2016-11-07:
Okay, i will do that. The test is postponed to wednesday. I hope i can get some results and the report is helping.

SearchForTheCode, 2016-11-09:
indeed, if i disable the mainboard the error is gone, because no access to the fan and other mainboard sensors is done.. So this helps but then i cannot access any temperature or other value of the mainboard.

I attached two reports and named them after the case. Hope this helps in some way

moellerm, 2016-11-09:
If you can test with the CPUID HWMonitor as well (and save/attach a report create with that tool), then this might help.

SearchForTheCode, 2016-11-09:
Okay just to be clear. I start the OHM version (causing the problem). Then closes OHM and start CPUID HWMonitir and create a report. Then i do a fresh reboot and start CPUID HWMonitor to create a "positive" report. 

I could do that, but it will take another 1 or 2 weeks in order to get access to the system.

SearchForTheCode, 2016-11-25:
I am kinda late but i have now a date and will try to get the information on 13.12.

SearchForTheCode, 2016-12-13:
So as promised i got the data and can describe more to it.

So my workflow was:

- Run OHM and create the FAN issue. 
- i started HWMonitor and it DID Not fix the FAN issue, the reports for this are name _afterOHM
-Then i did a fresh reboot 
-I started just HWMonitor and it did not create any FAN issue. These reports are called _afterFreshBoot.

So hopefully Michael you can get something from it :) 

[hwm_smbus_afterFreshBoot.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/648156/hwm_smbus_afterFreshBoot.txt)
[hwm_smbus_afterOHM.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/648157/hwm_smbus_afterOHM.txt)
[HWMonitor_afterFreshReboot.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/648154/HWMonitor_afterFreshReboot.txt)
[HWMonitor_afterOHM.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/648155/HWMonitor_afterOHM.txt)

SearchForTheCode, 2017-01-23:
Could you find out anything from the data?

Slion, 2017-05-05:
I had a similar issue, see #779.

SearchForTheCode, 2018-08-06:
I guess the data did not help?

DarthDestroyer, 2022-02-11:
There are some problems, if you use two apps or more, which access specific Hardware.
If you run only one, is it the same behavior?

btw. OHM can control the fan speed, did you set a value there?

kamenminkov, 2022-02-11:
I'm almost 100% sure that there are no issues if I only run AISuite.

From looking around the issues here it seems like my hardware isn't supported for fan control - I only get readings - so I think it should not interfere, and yet it seems like it does. The motherboard is an Asus Crosshair VIII Dark Hero (the controller comes up as "Nuvoton NCT6798D") and the CPU is an AMD Ryzen 9 5900X. The last time I had an issue it was the CPU fan that was stuck to 300-400 RPM (i.e. AISuite sensing that it should ramp up the fan, but not actually ramping it up), so the CPU almost hit 100 degrees C (and at the same time the case fans, also hooked up to the motherboard, were responding properly).

I'll be happy to provide any other info needed to debug this.

DarthDestroyer, 2022-02-11:
Could you try it without the asus Software and only OHM, if it then works properly?

kamenminkov, 2022-02-11:
Just tried and it seems to do. But again that was some kind of an edge case that I can't yet isolate (i.e. I'm not sure how to reproduce it).

My bad about the fan control - it's there, I just had forgotten that it's not in the main menus but in the context menu of the table.

What does the "Default" option for fan control mean? That the fan falls back to its underlying control? Once I check it, I can't uncheck it, and I don't know if it makes a difference (i.e. "Default" checked vs nothing checked).

DarthDestroyer, 2022-02-11:
to be honest, i would have to look at the code.
I never really used that part